aredn/aredn_ar71xx #322: HAP AC Lite Enhancement Request Suggestions

k1ky, 2019-01-01:
Suggested enhancements to access the dual-band features of the HAP AC Lite unit:

1. Enable both Ports 2.4 and 5 Ghz as MESH nodes in order to allow MESH connections to both 2.4 and 5GHz units.
2. Reverse the function of the 2.4 and 5GHz ports - i.e. 5 Ghz MESH and 2.4 Ghz AP
3. Independent power control in AP mode as well as signal strength monitoring and Wi-Fi scanning
4. Bridge mode in Wi-Fi in order to provide internet to a MESH node from a Wi-Fi system.

k1ky, 2019-01-01:
Of course applicable toward single-band devices where possible to include Power control, Wi-Fi scan, SNR Charts and History & Bridge mode in non-MESH Wi-Fi mode.

Orv, 2019-01-01:
I like this idea, but if both radios in the HAP AC Lite are configured as mesh nodes, they'll need different hostnames & IP addresses.   Configuration would require significant reworking of the UI, I'd think, as you'd now have two of everything to configure:  channels, bandwidths, distance settings, etc.

k1ky, 2019-01-01:
Yes, certainly an whole new ballgame - and of course a lot of unknowns regarding what may happen with loading factors and the like.  But as hardware capabilities expand....  baby steps perhaps?  Just thought I would put it out there!

Orv, 2019-01-01:
I'm all for it, because eventually we'll have some sort of a node with > 1 mesh radios in it; Mikrotik Routerboards support mini-PCI-based radios and I could see AREDN firmware running some sort of Routerboard with three or four radios on it - that would be all you'd need at a site for full-networking.   (No built-in anennas, but you can't have everything...)       That's assuming appropriate drivers ever became available...

k1ky, 2019-01-01:
After reviewing - it may be probably best to break this up into "bite sized" chunks, creating multiple feature requests??

dman776, 2019-01-01:
My thoughts exactly

Thanks,
Darryl

k1ky, 2019-01-01:
And of course there is that pesky USB port with all sorts of possibilities! RF boards, more storage, etc. But of course functions that may be more suited for RPi devices.

n2mh, 2019-01-02:
I'm not sure that multiple radio ports (interfaces) would require multiple node names. Multiple ip addresses, yes, but not multiple names. After all, every node out there today has multiple ip addresses that all resolve back to the same node name: LAN port, RF port. And, if there are tunnels, multiple addresses for each tunnel connection (interface) (172.31.x.y). But, they all resolve back to the one node name.

ae6xe, 2019-01-02:
This would not be a bridged interface with a single IP address, since this would put all traffic on both bands.  Traffic should only be on the specific band the destination takes it.    Historically, we have propagated hostnames for the dtdlink interfaces, but hide these in mesh status in place of the primary node hostname.  I'll have to review, we may not need to have this dtdlink host name and double entry for every node and might be able to cut down the size of the OLSR data propagating around.

When there is a setting of 2 APs or 2 Mesh RF, then the UI layout has to be thought out.

dman776, 2019-01-10:
> A few more future feature enhancement ideas:
> 5. USB WAN interface capability - to be enable Cellphone, Cellular Hot-Spot and Wi-Fi devices to supply internet.
> 6. USB Data storage support - file server

These need to be created as separate issues

Thanks,
Darryl

k1ky, 2019-01-14:
Will do - discussed with Randy and I will create separate tickets for bite-sized chunks.