aredn/aredn_ar71xx #323: MikroTik hAP Lite Advertise Services

denis133, 2019-01-05:
Build #600 Whenever node powers up any advertised services must use the Save Changes to link to the service.

ae6xe, 2019-01-06:
Please clarify, I'm not understanding what the problem or failure is.    You do not see services on the mesh status page that are expected?   (and had previously defined and 'saved' these definitions.)   Screen shots, step-by-step to reproduce are helpful.

denis133, 2019-01-06:
Sorry Joe. Correct, I do not see the clickable link on the status page until I go to the Service page and save the changes when in fact it is already there. One must tap the "Save Changes" button to re-engage the service that is already configured previously.

ae6xe, 2019-01-06:
Ok,  reboot node, capture a data download (before condition),   "save changes", and then capture a 2nd data download (after condition).  This will hopefully provide sufficient details to get to the bottom of the issue.

denis133, 2019-01-07:
First file after reboot:
[supportdata-KD1HA-BUR-201901062004.zip](https://github.com/aredn/aredn_ar71xx/files/2731099/supportdata-KD1HA-BUR-201901062004.zip)
2ed after "Save Changes"
[supportdata-KD1HA-BUR-201901062006.zip](https://github.com/aredn/aredn_ar71xx/files/2731100/supportdata-KD1HA-BUR-201901062006.zip)

ae6xe, 2019-01-07:
The service advertisement is saving correctly.   What is not happening, is OLSR is missing this service's hostname definition.   I suspect root cause is the same as  https://github.com/aredn/aredn_ar71xx/issues/204 .

You should be able to get back to normal by performing the same work around (which is executed with the services save settings button too):

/etc/init.d/olsrd restart 

As a test, can you type this one-time command to change the timing of when OLSR starts up at boot, to see if it resolves your issue?  this change is permanent across boots:

mv /etc/rc.d/S65olsrd /etc/rc.d/S97olsrd

This will delay the startup of olsr and may address the root cause, if it is a timing issue as I suspect.  See if you can reproduce the symptoms after this change and reboot.

denis133, 2019-01-07:
After adding the above line mv /etc/rc.d/S65olsrd /etc/rc.d/S97olsrd does delay startup for about 5-6 seconds. I can not reproduce the failed symptom as previously described. The link now appears and one can link to the service. Thank you.

Denis KD1HA

denis133, 2019-01-07:
After adding the above line mv /etc/rc.d/S65olsrd /etc/rc.d/S97olsrd does delay startup for about 15-16 seconds was thinking that was a bit too long and maybe we had another issue but it started fine. I can not reproduce the failed symptom as previously described. The link now appears and one can link to the service. Thank you.

denis133, 2019-01-08:
After adding the above line mv /etc/rc.d/S65olsrd /etc/rc.d/S97olsrd does delay startup for about 15-16 seconds, thinking that was a bit too long and maybe we had another issue but it started fine. I can not reproduce the failed symptom as previously described. The link now appears and one can link to the service. Thank you.

ae6xe, 2019-01-11:
change committed:  https://github.com/aredn/aredn_packages/commit/4fe5e8eec06f7900b71a862c645ea374467c7fcd

closed 2019-01-11 by ae6xe