aredn/aredn_ar71xx #331: Reboot still required for sysupgrade

Orv, 2019-01-23:
It appears that a reboot is still frequently required prior to a sysupgrade, even on a 64 MB device, if it's been running for more than a few days.   Advise if any research is needed.

Orv, 2019-02-03:
Still may be an available RAM issue as upgrades appear to work the first time if a reboot is done prior.

Orv, 2019-02-12:
An observation - after upgrading to nightly build 705 on about 10 nodes:

- Success rate when upgrading nodes running the production release is greatly improved.  
- Success rate when upgrading nodes running nightly build 692 is almost always successful.

Orv, 2019-02-19:
Another factoid:   Attempted to upgrade a Rocket M3 from 3.18.9.0 to latest nightly build (713).  It had been up 100 days.   The binary file uploaded, the node went to the "upgrading firmware, please wait" page, and then eventually timed out.  When it was accessible again, it had rebooted, but NOT upgraded.
The second attempt, shortly after it rebooting, was successful.

raysuelzer, 2019-06-02:
This could be related to my issue of being unable to sysupgrade Mikrotik LHH 5hpnd

ae6xe, 2019-08-16:
https://github.com/aredn/aredn_ar71xx/commit/e316dbc598163fa5258a3fb748e8ff5775f10356

After applying this fix, I have not seen a sysupgrade failure.  One example, I sysupgraded a NSM3 with uptime of over a month with success.  

Everyone, please post your results to determine if this issue can be closed.

Orv, 2019-08-16:
> e316dbc

In which nightly build was it applied, Joe?

Orv

Orv, 2019-08-16:
Never mind - saw your post in the Forum.  I have lots of nodes I took to 
1000, and then beyond, with no issues.  Yes, Resolved, IMO.

Orv

ae6xe, 2019-08-16:
build #1000 had later commits on June 20th.   The first build with this fix would be the one just prior to that, out on June 19th.

ai6bx, 2019-08-24:
I have recently updated with 1022 on a number of devices ranging from Nanobridges, Rockets, and PBE. The longest running device was an older Nanobridge that was on 3.19 and had been going for 68 days. All seemed to take 1022 without incident.

closed 2019-08-24 by ae6xe