aredn/aredn_ar71xx #333: Typo in additional terms of license on website

jamespbryan, 2019-01-23:
Hello,

This is not a code issue, but the typo might appear in other areas and certainly in downstream projects. On the site:
https://www.arednmesh.org/content/ui-license
the line below should likely be changed. Recommended changes are in bold:
=========ORIGINAL==========
Modified versions must be modified to attribute **to** the original source and be marked in reasonable ways as differentiate it from the original version.
==========================

=======RECOMMENDED========
Modified versions must be modified to attribute the original source and be marked in reasonable ways as **to** differentiate it from the original version.
==========================

jamespbryan, 2019-01-23:
This appears to show up in more than just the website. I found it in the project source's license file as well last modified in 2018.

ae6xe, 2019-01-26:
Website wording corrected.

closed 2019-01-26 by ae6xe