aredn/aredn_ar71xx #334: Another Typo in additional terms of license on website

jamespbryan, 2019-01-23:
Hello,

This is not a code issue, but the typo might appear in other areas and certainly in downstream projects. On the site:
https://www.arednmesh.org/content/ui-license
the line below should likely be changed. Recommended changes are in bold:

ORIGINAL

> The AREDN™ trademark and AREDN™ logo **and** for use in official Amateur Radio Emergency Data Network (AREDN™) sanctioned builds.

RECOMMENDED

> The AREDN™ trademark and AREDN™ logo for use in official Amateur Radio Emergency Data Network (AREDN™) sanctioned builds.

**The summarized change is the removal of the word "and".

Thanks!

wu2s, 2019-01-26:
Thank you for catching this.

wu2s, 2019-01-26:
The recommended change is approved.

ae6xe, 2019-01-26:
web site is updated.

closed 2019-01-26 by ae6xe