aredn/aredn_ar71xx #344: Package Management removal messages - Add indicator when finished?

k1ky, 2019-01-29:
Requesting clearer messaging when removing packages in the Administration area to include a message indicating when the procedure is "done" or complete and it is safe to move on, reboot or whatever.  When installing packages there is usually some indication when that procedure is complete with instructions on what to do next.  

Shouldn't there be something confirming that the removal of a package is complete in order to avoid any confusion?

dman776, 2019-01-30:
If the package uninstall returns any text, we display it on the GUI.   If nothing is returned from the package uninstall routine, we have nothing to display.   Are you asking that we synthesize a response *IF* the package doesn't return anything to us?

k1ky, 2019-01-31:
Yes - some indication that the processs/action has been completed and it is "safe" to move on.  Is there any way to test for that and report?  Otherwise - it may be prudent to request a code fix in the packages - of which my request would be for the VTUN and MESHCHAT modules have code added to confirm removal.

dman776, 2019-01-31:
Is the browser’s “loading” indicator not enough?
Ie.  The spinning circle on the tabs when the page is processing/loading.

k1ky, 2019-02-03:
Per Darryl's comment: Ie.  The spinning circle on the tabs when the page is processing/loading.
So is it "safe" to assume that when the circle in the tab quits spinning that the removal is complete??
I don't believe I have experienced any problems after removal, but I do get questions from our users when packages are removed - seems that it would be good to receive a "confirmation" that the process didn't hang up somewhere - i.e. if it was a sluggish node.

dman776, 2019-03-27:
I'm transferring this issue (since it is based on the Perl UI today) to the New AREDN UI (Angular) repo for capture