aredn/aredn_ar71xx #355: Feature Request- Add Node to Node push message feature within native  AREDN FW

ZL2TB, 2019-02-06:
Propose a simple push (i.e. when sent the  msg pops up on selected node if PC logged into that node or directly connected to it monitoring MESH ) message button is added native to MESH FW to send short msgs to a selected node over the MESH. A basic engineering communications channel. As simple as possible to save code space. Could provide last resort comms via the MESH if no node with "meshchat" is available too.

k1ky, 2019-02-07:
Interesting concept - sort of a "mini packet" mode.

ZL2TB, 2019-02-07:
I imagined it may work like this, a Message virtual  button is pressed and a drop down list of connected nodes appears , you select one and then enter text into a dialogue box  and "send" when complete. The Message appears at the specified node if a PC is active on the node or is buffered until one is. Only session history would be kept while application is active and messages are lost when app is closed for space reasons. If space was not an issue more bells and whistles would be great

k1ky, 2019-02-07:
I remember in the early days that we used an application module that we installed on each node "hamchat" which was a non-synchronized chat room on each node - maybe some simple implementation of that, but then again - more overhead!