aredn/aredn_ar71xx #358: Nanobeam Untested & no amateur channels

rfgarvey, 2019-02-07:
I have a nanobeam NBE-M2-13. Finally got the AREDN firmware to flash and I have the current nightly build. Localmesh web page for my device says 'Hardware Untested' and in the setup, there are only the commercial channels 1-11.
I have attached screen captures.
Thanks.

[N3KAL_nanobeam_mesh_hardware report.pdf](https://github.com/aredn/aredn_ar71xx/files/2839352/N3KAL_nanobeam_mesh_hardware.report.pdf)
[N3KAL_nanobeam_mesh_hardware report_2.pdf](https://github.com/aredn/aredn_ar71xx/files/2839353/N3KAL_nanobeam_mesh_hardware.report_2.pdf)
[N3KAL_nanobeam_mesh_hardware report_3.pdf](https://github.com/aredn/aredn_ar71xx/files/2839354/N3KAL_nanobeam_mesh_hardware.report_3.pdf)

ae6xe, 2019-02-07:
Please capture the data support download file, bottom of Administration page, and post back here.

rfgarvey, 2019-02-07:
Thank you for looking into this.
This was captured from a linux laptop.
[supportdata.zip](https://github.com/aredn/aredn_ar71xx/files/2842155/supportdata.zip)

ae6xe, 2019-02-07:
rfgarvey,  this device (hardware revision) has not been seen yet.   Is it possible to reload AirOS and capture the "/etc/board.info" file?   We need to see the manufacture power settings to determine how to configure in AREDN.

rfgarvey, 2019-02-08:
I'll do my best. The only thing I have to go on is the original AirOS was v5.6.11 (XW). I didn't write down anything else at the time.
Rob

rfgarvey, 2019-02-08:
I just checked the Ubiquiti site... v5.6.15 may be the oldest I can download.
Any thoughts?

ae6xe, 2019-02-08:
You can download a current AirOS version and tftp upload.  Then afterwards, tftp back to AREDN firmware.    The downgrade process is no longer applicable with 3.18.9.0 and nightly builds.  When AirOS is loaded, telnet to 192.168.1.20 and login with user "ubnt" and password "ubnt".  the "cat /etc/board.info".

rfgarvey, 2019-02-08:
Okay, I already grabbed the v5.6.15. Do you want me to grab the newest instead? (I have not rolled back the firmware yet)
Rob

ae6xe, 2019-02-08:
Any version you upload of AirOS is fine.

rfgarvey, 2019-02-08:
under the aredn software, what is the IP address for tftp when I do the flash of the Airos? is it still 192.168.1.20?

ae6xe, 2019-02-08:
When the device is in tftp mode and when AirOS is running, access with 192.168.1.20 (static).  When the device has booted AREDN for the firstboot, it is 192.168.1.1 (dhcp)

rfgarvey, 2019-02-08:
thanks. i ready to roll it back with the latest airos.  hopefully i'll be 'right back'  ;)

rfgarvey, 2019-02-08:
the device is in the 1-3 2-4 state ready to flash. i cannot ping 192.168.1.20. arp -a at the command line shows no other devices. it still has the aredn firmware on it.

what should my tftp command look like?
I am thinking: tftp -i 192.168.1.20 PUT XW.v6.1.9.32918.190108.1737.bin

ae6xe, 2019-02-08:
if you can not ping 192.168.1.20, no need to proceed, something is not right.   I always use a dumb switch between the laptop and the device.  I've found they don't respond unless there is a link up detected on the ethernet port when it powers on.

rfgarvey, 2019-02-08:
i just connected through a spare router i had.  looks like the flash is taking.  i'll let you know in a few minutes

rfgarvey, 2019-02-08:
airos v6.1.9 (xw) confirmed through browser
telnet fails on port 23

ae6xe, 2019-02-08:
might have to ssh in on port 22,  that's what I always use.  Probably telnet is not running.

rfgarvey, 2019-02-08:
okay. got in with ssh. did a cat on board.info and copy/pasted to a txt file. this is on a separate machine than this one so i will have to sneaker-net it via usb stick to paste here.

 do you need anything else while i'm in there?

ae6xe, 2019-02-08:
That's it.  the rest we have already from the AREDN support dump.

rfgarvey, 2019-02-08:
[n3kal_nanbeam_airos_boardinfo.txt](https://github.com/aredn/aredn_ar71xx/files/2843506/n3kal_nanbeam_airos_boardinfo.txt)

rfgarvey, 2019-02-08:
thanks for your help on this!

ae6xe, 2019-02-08:
I'll commit the changes and should be in the images built tonight.     Check tomorrow and test.

rfgarvey, 2019-02-08:
Wow! Thank you!

PR #359 (2019-02-08): aredn: add definition for nanobridge M2 13
fixes #358
commit 711fe96a (2019-02-08): aredn: add definition for nanobridge M2 13

fixes #358
closed 2019-02-08 by ae6xe