aredn/aredn_ar71xx #360: NanoBeam Untested Hardware issue 358 - no change

rfgarvey, 2019-02-08:
I downloaded and flashed last nights build. There is no change.
The file I am using is: aredn-692-bb818a9-ubnt-loco-m-xw-factory.bin

[n3kal-nanobeam-untested-1.pdf](https://github.com/aredn/aredn_ar71xx/files/2845406/n3kal-nanobeam-untested-1.pdf)
[n3kal-nanobeam-untested-2.pdf](https://github.com/aredn/aredn_ar71xx/files/2845407/n3kal-nanobeam-untested-2.pdf)
[n3kal_nanbeam_airos_boardinfo.txt](https://github.com/aredn/aredn_ar71xx/files/2845408/n3kal_nanbeam_airos_boardinfo.txt)
[supportdata-N3KAL-NanoBeam.zip](https://github.com/aredn/aredn_ar71xx/files/2845409/supportdata-N3KAL-NanoBeam.zip)

ae6xe, 2019-02-08:
OK, one more change will be needed.  1 step forward, should be just one more to go.   We may need to do a couple of iterations offline.  Can you email me at ae6xe at arrl dot net (so I have an address to reply to)?  I'm traveling today, so this weekend.

rfgarvey, 2019-02-12:
Thanks you Joe ae6xe for taking care of this!!

closed 2019-02-12 by rfgarvey

ae6xe, 2019-02-12:
change had been successful.  Build failed on CircleCI nightly build due to other reasons and was addressed.   Used too many parallel cpu's, and sometimes Openwrt builds don't complete and have to be run with less cpu's, due to timing and dependencies getting out of order.