aredn/aredn_ar71xx #362: Firmware upgrade page reload not functional

Orv, 2019-02-12:
In nightly build 705, the page reload function on the firmware upgrade page (states page will reload in 3 minutes) doesn't appear to work - after sitting there over 7-8 minutes (didn't time it exactly), the browser disconnects from the node.  Happened several times, but will do more testing and report back.

ae6xe, 2019-08-25:
Orv, been a while on this issue.  Have you noticed any change in these symptoms?   The web page is using an html meta delayed redirect.    This could be a symptom of add-blockers or other anti-virus tools.   Adware exploits such redirects...

Here's one example of what can prevent the redirect from working.

ref:  https://stackoverflow.com/questions/522353/meta-refresh-doesnt-work :

"There is a security setting in internet explorer that does not allow meta tag refresh. It is under the Security tab, then choose Custom Level and the Meta Tag Refresh under Miscellaneous. If that is disabled, it would stop the meta refresh from working.

Aside from being able to disable it selectively, it is automatically disabled if you set IE's security level to 'High' [observed in IE9]."

Orv, 2019-08-27:
I'll review this when I return. I'm traveling for the next week or so and
will be on and off the grid while gone.

ae6xe, 2020-04-22:
bump, considering closing this this.

Orv, 2020-04-22:
It still doesn't work, but it's not highly critical.  You can close it if you want.

ae6xe, 2020-08-06:
I'm going to go ahead and close this.  There have been posts in openwrt on this subject and there is no guarantee the sysupgrade process won't hang.  This is due to the fact that to add features in the kernel to make this guarantee would bloat the kernel and a bigger negative.  Openwrt developers had determined it was a better trade off to continue to allow the very small/tiny probability that the sysupgrade process may hang.  Note, however, the issue of a device with only 32M of RAM that may cause this symptom is a different root cause and not something to classify as a defect per se.

closed 2020-08-06 by ae6xe