aredn/aredn_ar71xx #363: Enhancement: Provide additional flexibility in the entry of links for advertised services

n2mh, 2019-02-13:
Recently, I was working with a station to enter a link to his Foscam TV camera but found that the entry boxes on the Port Forwarding config page to do so could not handle the needed url:

`rtsp://view:view@10.1.2.3:443`  

In particular, the "view:view@" could not be entered.

Please look into this and see what can be done. I believe this issue goes back to Day 1.
Thanks!

r1de, 2019-02-13:
Hmm...  Maybe cause that is *not* how PORT forwarding works.

ae6xe, 2019-02-13:
Mark, you must be referring to the service advertisement on the port forwarding.... page.   There are some limitations, e.g. what OLSR allows when propagating around the mesh.  here's the check on the suffix after the ":".

  # checks if string meets critera specified by nameservice module
  return 0 unless $suffix =~ /^[[:alnum:]\/?._=#-]*$/;

but you are referring to limitations in a prefix of the hostname, which isn't handled in the code -- just includes the prefix as part of the check for a valid hostname.   This could be allowed, as long as OLSR limitations support the characters.  have to look a little more closely.

n2mh, 2019-02-15:
Joe, Yes, that is what I am referring to - the right side of the config page where one optionally assigns a link to a reserved address. 

FWIW, Amcrest also has the same format for their rtsp:// url:
<a href="rtsp://view:view@towercam.local.mesh:554/cam/realmonitor?channel=1&subtype=0">Camera</a>

Just for giggles, I tried manually configuring that url into a PlParam "services" line in olsrd.conf and it did not work. It appeared to like the @ sign but not the colon (:). I even tried faking out the colon with a %3a but that didn't work either.

So, unless you can find something else Joe, I guess this type of url is not possible in the OLSR protocol. Perhaps a workaround would be to establish a web page somewhere that could be advertised with an acceptable url with a redirect once someone gets there.

Thanks for looking into this.

ae6xe, 2019-02-20:
Mark, it is tricky to make the change in the /tmp/run/olsrd.conf file directly.  It gets re-written when you stop/start olsrd.     Did you do this in /etc/config/services first?    Then "/etc/init.d/olsrd restart" to find the expected entries in /tmp/run/olsrd.conf (the generated run-time config file)?