aredn/aredn_ar71xx #364: WAN access not functional through Mikrotik

Orv, 2019-02-17:
5 GHz and 2 GHz nodes on tower get their Internet access (when enabled), through a Mikrotik RB952Ui-5ac2nD.  I believe it had worked when I initially installed the Mikrotik.   Recently I discovered that when the "Allow others to use my WAN" check box is checked, neither node gets a default gateway from the Mikrotik.

Running root@W6BI-Shack-Node:~# ip rule list | grep "lookup 31" yields

20080:  from all iif eth1.2 lookup 31
20080:  from all iif eth1.3975 lookup 31
20080:  from all iif tun50 lookup 31
30099:  from all iif br-lan lookup 31
30299:  from all lookup 31

I upgraded the Mikrotik to the lasted nightly build, but still no joy.  
Am I doing something wrong?
[supportdata-W6BI-Shack-Node-201902162058.gz](https://github.com/aredn/aredn_ar71xx/files/2872675/supportdata-W6BI-Shack-Node-201902162058.gz)
(renamed the support data file from .tgz to .gz to upload)

ae6xe, 2019-02-22:
Orv, this is from build #509, very old.  Can you upgrade this to current nightly and retest?  I'm unable to reproduce on currently nightly.

Orv, 2019-02-22:
Here's on with it running 713 (behavior's the same)
[supportdata-W6BI-Shack-Node-201902220831.gz](https://github.com/aredn/aredn_ar71xx/files/2894902/supportdata-W6BI-Shack-Node-201902220831.gz)
(renamed the support data file from .tgz to .gz in order to upload).

ae6xe, 2019-02-22:
This looks correct to me on this node (for traffic coming in on the dtdlink).  It would get to the local internet.  the relevant rules for traffic incoming on dtdlink (interface eth1.2):

20010:	from all iif eth1.2 lookup 29  <- routes to the node's LAN
20020:	from all iif eth1.2 lookup 30  <- routes to hosts on the mesh
20080:	from all iif eth1.2 lookup 31   <- routes to mesh gateways -- this is empty because self node is a gateway
20090:	from all iif eth1.2 lookup main  <- default route to local internet

The first route that is applicable and exists, going down this list, is the one in the main table and would route traffic incoming on the dtdlink to the WAN.  This is the local main table default route.

Need a support dump on a node that can't reach the internet to see what is happening on that side of the fence.

Orv, 2019-02-22:
Here's an NSM2, DtD connected:

[supportdata-W6BI-VC-QTH-2G-201902221015.gz](https://github.com/aredn/aredn_ar71xx/files/2895200/supportdata-W6BI-VC-QTH-2G-201902221015.gz)

ae6xe, 2019-02-23:
I do see some defects in the configuration on this node.  However, it too is an old 536 build.  Please update to latest and test one more time and send support data as appropriate.  In comparison to a NanoStation XM device I have, the same issues are not there.  However, it's still possible you have some different combination of settings, so let's see what happens with the update.

Orv, 2019-02-23:
[supportdata-W6BI-VC-QTH-5G-201902221708.gz](https://github.com/aredn/aredn_ar71xx/files/2896316/supportdata-W6BI-VC-QTH-5G-201902221708.gz)

Here you go - a different node.  However, be advised that NO nodes see my Mikrotik's path to the Internet.  So the issue probably doesn't lie with them.

Orv, 2019-02-24:
Closing - issue identified.

closed 2019-02-24 by Orv