aredn/aredn_ar71xx #365: TUNNEL Connection Does not "Re-acquire" after reboot

k1ky, 2019-02-20:
Tunnel connection does not automatically "reconnect" after reboot of Client unit Nightly 713.  Tunnel server still has a "blue cloud" even though the links are not shown and there is no blue cloud on the rebooted tunnel client side.  Must either reboot the tunnel server or disable/re-enable server port for that node to reconnect. 

There should be some sort of "activity" checking and automatic restart procedure for non-functioning tunnel connections on the server end.

This particular case is a UBNT AirRouter Client Nightly 713 and Tunnel Server Microtik hAP AC Lite Nightly 705.  This has been a problem for quite some time and is not a "new" issue.

ae6xe, 2019-02-20:
Is it possible to obtain support dumps from both sides of the fence and upload?

k1ky, 2019-02-20:
Yes, I will endeavor to get that information for you.

k1ky, 2019-04-22:
Still experiencing this issue - observed with HAP AC Lite on both ends -client unit running production, the server was running an earlier nightly version.  The tunnel still had a Blue Cloud and was answering Putty SSH calls from the problem Client end but would not connect the far end (server) node via tunnel.  There were 4 other clients still connecting and functional.  Have upgraded to Production 3.19.3.0 on all units after this occurred - will observe and gather Support dumps when it happens again.  Please keep this issue alive as we gather more data.

k6ccc, 2019-05-02:
I observed this last week as well.  Made a minor change on the config of the client end that resulted in a reboot of the client.  The tunnel did not come back up.  As I recall, disabling the tunnel and re-enabling it brought it back up.  Fortunately I had left a PC connected to the client that I could access via TeamViewer to make the change.  Both ends were Rocket M5 radios with 3.19.3.0 firmware.  I will plan on setting up the laptop again this evening so I can attempt to replicate the condition and get the support dumps.

k6ccc, 2019-05-03:
Figures.  Tried it again this evening and the tunnel re-connected every time without any intervention...

k1ky, 2019-05-03:
Suspecting that the Tunnel Server isn't "letting go" of the connection after the client connection is disrupted.  Maybe there could be some connection verification watchdog procedure that can monitor the status of tunnel connections and reset the connection if trouble is detected on the server end.  There may already be something like this in the firmware and we just aren't giving it enough time to reset.

k6ccc, 2019-05-03:
At least the tunnel active indication on the server showed the tunnel inactive.  I have no idea if there is something on the back end that was still hung.  I was quite surprised last night when I tried it, that the tunnel active indication on the server went to inactive almost immediately when I rebooted the client end.  I did not try a reboot of the server end.

I did just try a reboot of the server, and the tunnel did not connect.  However it's not related.  The hAP at home is in a continuous reboot state.  I saw this once before.  Not practical to walk my wife through troubleshooting that so it will have to wait until I get home this evening.

k1ky, 2019-06-18:
Running TOP on the server node shows active connection from the client even after rebooting the server.   I have obtained support dump from the server end.  Working on getting a support dump from the client once I gain access.

k1ky, 2019-06-20:
Support files e-mailed to AE6XE June 19 @ 19:30 CDT