aredn/aredn_ar71xx #378: Automatically Re-install Tunnel and keep Settings

denis133, 2019-03-16:
Feature request: Automatically re-install a tunnel and keep settings to a remote node on a tunnel server/client after firmware update. We need to have someone at both ends to simply update firmware on a working tunnel or travel to the remote site to re-establish the tunnel. 
We have remote nodes not easily accessible in secure places.  

Denis KD1HA

k1ky, 2019-03-17:
I 2nd that notion - this is something that I have intended to request.   We have many remote tunnel sites that would benefit from this enhancement feature.

ae6xe, 2019-03-17:
Since the device would have to have internet access to re-establish the tunnel, it is possible to setup a port forward to 80 or 8080 of the node.  It can be reached from the internet to install the tunnel after the upgrade.   Then close down the port forward to protect.

k1ky, 2019-03-17:
A lot of the units we would want to upgrade aren't setup for "incoming" access, but that most certainly could be an option.  I have some units with a computer attached to them and I remote into the computer to complete the tunnel upgrade. I believe what you are proposing would require opening up a port forward in the client's home router?