aredn/aredn_ar71xx #385: Minor typo in 3.19.3.0 release notes

Orv, 2019-03-23:
Minor typo in "AREDN Release Notes v3.19.3.0 – 23 March 2019"

Is:  A new feature adds sounds, including pitch and volume control, to the real-time SNR chart page. As a device is tilted or rotated towards a remote node, the computer generated tone changes pitch based on a higher of lower SNR. This feature assists in aiming a node by indicating a better signal with a higher tone.

Should be:   A new feature adds sounds, including pitch and volume control, to the real-time SNR chart page. As a device is tilted or rotated towards a remote node, the computer generated tone changes pitch based on a higher **or** lower SNR. This feature assists in aiming a node by indicating a better signal with a higher tone.

dman776, 2019-03-23:
corrected.  thank you.

closed 2019-03-23 by dman776

Orv, 2019-03-23:
You're very welcome. Great work, guys!