aredn/aredn_ar71xx #393: Preventing LAN devices from accessing WAN doesn't work completely

n4tqu, 2019-03-27:
I upgraded to 3.19.3.0 today and decided to try the option for "Prevent LAN devices from accessing WAN" and noticed that this does not work as written. An existing LAN device will continue to access the WAN at least until the DHCP lease on the client renews and drops the default route. Also, anyone can add a static route on the LAN device and be capable of accessing the WAN.

I suggest an iptables rule would likely be more appropriate as a preventative measure.

Additionally, checking this box really should not require a restart of the mesh node. I suspect this is done due to limitations of the hardware however.

Steve - N4TQU

ae6xe, 2019-03-27:
This option was previously called "Disable default route". The wording change was an attempt to clarify the intent for the non-IT guru.  The original implementation (for years) has not changed.    This new wording certainly communicates restrictions that are not imposed.    We have 2 options to proceed; A) change the wording to better match what is occurring;  B) add the logic that  prevents manual circumvention to access from the LAN to a WAN (local or remote).    Feedback, preferences anyone?

dman776, 2019-03-28:
I thinks it’s fairly clear in the docs.
https://arednmesh.readthedocs.io/en/stable/arednGettingStarted/advanced_config.html

Perhaps a wording change would help.

Thanks,
Darryl

n4tqu, 2019-03-29:
> This option was previously called "Disable default route". The wording change was an attempt to clarify the intent for the non-IT guru. The original implementation (for years) has not changed. This [...]

My vote would be for real prevention. I think this is especially important for those just getting into AREDN and utilizing tunnels to explore the capabilities of the system.

ae6xe, 2019-03-29:
There is a use case where the LAN devices might be adhoc users on mobile device with a voip app.   The general scenario is users that are not the node admin.    Controlling WAN access would be a desired capability IMO. Another node might give a group of users access to the WAN.  Thus it's not an "all or nothing" option for WAN access to be able to manage a limited resource to those with the most need.  The prevention would build in security so a savvy user could not circumvent what is intended.

n4tqu, 2019-03-29:
Taking this idea a step further one could implement an option to grant a per MAC address reservation (LAN device) with WAN access.

K6AH, 2019-04-17:
I'd be in favor of renaming it back to Disable Default Route and use the documentation to explain what that really means.

r1de, 2019-04-18:
I'm with Andre on this one, if all the checkbox does is cause dnsmasq to not give out a default route with the leases then name it back to "Disable Default Route".
I guess that would also mean, we'll need a new checkbox then that turns on/off a new firewall rule.