aredn/aredn_ar71xx #397: Delete non-existant ssh key kills dropbear

ke4ahr, 2019-04-03:
Login to the admin pages, hit the button to select a ssh public key, select a key... then hit the button to delete the key instead.

Kills dropbear, any existing ssh session, disrupts vtun connections. Repeatable, affects both current and previous releases.

current version: 3.19.3.0 hardware type: ubnt (bullet-m)

dman776, 2019-04-17:
Can you record your screen (video) and link or upload it here?  I can't seem to replicate this.

ke4ahr, 2019-04-18:
Try using Firefox instead of Chrome. I will see if I can capture the queries with a proxy.

dman776, 2019-04-19:
I was testing on Firefox 66.0.3 (64-bit) for OSX.

r1de, 2019-04-30:
I could not recreate this either.  
Using Firefox on Linux.  

I did have my ssh session terminated when I reuploaded my cert (after testing this and deleting it), but there were no other disruptions other than that.

dman776, 2019-05-06:
closing.   not able to replicate and no additional info from reportee.

closed 2019-05-06 by dman776