aredn/aredn_ar71xx #398: Ubiquity Litebeam M5 table of defaults not being used?

mpannen1979, 2019-04-07:
Max power, Model name not being used for power selection and status page.  The status page lists:

model: Unknown Hardware 

!!!! UNSUPPORTED DEVICE !!!!
boardid: Ubiquiti Litebeam M5

and I can select a max power higher than 19, even though the perlfunc file says supported=1, and the maxpower option is defined to 19.

Maybe I don't understand how the max power selection works--I am new to this project.

When I change maxpower to something higher, save, and reboot, it seems the device 'reboots' at least twice, and the power ends up back at 19dbm.

Are the table values not applying somehow?

mpannen1979, 2019-04-07:
in perlfunc.pm, I changed this table 'model' from '0xe865' to 'Ubiquiti Litebeam M5' and made a test firmware, and things look 'better':

''0xe865' => {
      'name'            => 'LiteBeam M5',
      'comment'         => '',
      'supported'       => '1',
      'maxpower'        => '19',
      'pwroffset'       => '6',
      'usechains'       => 0,
      'rfband'          => '5800ubntus',
    },

'Ubiquiti Litebeam M5' => {
      'name'            => 'LiteBeam M5',
      'comment'         => '',
      'supported'       => '1',
      'maxpower'        => '19',
      'pwroffset'       => '6',
      'usechains'       => 0,
      'rfband'          => '5800ubntus',
    },

ae6xe, 2019-04-08:
This may be a new hardware revision.     to confirm, can you reload AirOS and capture the output of /etc/board.info ?    We'd need to confirm the power offset and max settings, possibly different.  It appears to be different than 0xe865.

mpannen1979, 2019-04-08:
I have 2 units, and both showed this weirdness.  They were 'tested' on 7/30/2017 and 7/25/2017.

This is from the 7/30/2017 unit:

board.sysid=0xe865
board.cpurevision=0x00001122
board.name=LiteBeam M5
board.shortname=LB5
board.subtype=
board.netmodes=7
board.hwaddr=788A2092BEFD
board.reboot=40
board.upgrade=150
board.phycount=1
board.phy.1.maxmtu=2024
board.phy.2.maxmtu=2024
radio.1.name=
radio.1.shortname=
radio.1.bus=ahb
radio.1.subsystemid=0xe865
radio.1.subvendorid=0x0777
radio.1.txpower.max=25
radio.1.txpower.min=-4
radio.1.txpower.offset=6
radio.1.ieee_modes=5
radio.1.antennas=2
radio.1.def_antenna=1
radio.1.antenna.1.id=1
radio.1.antenna.1.name=Feed only
radio.1.antenna.1.gain=3
radio.1.antenna.1.builtin=1
radio.1.antenna.1.chains=1
radio.1.antenna.1.chain.1.id=0
radio.1.antenna.1.chain.1.name=Vertical
radio.1.antenna.2.id=2
radio.1.antenna.2.name=11x14
radio.1.antenna.2.gain=23
radio.1.antenna.2.builtin=1
radio.1.antenna.2.chains=1
radio.1.antenna.2.chain.1.id=0
radio.1.antenna.2.chain.1.name=Vertical
radio.1.distance_limit_km=0
radio.1.chanbw="5,8,10,20,30,40"
feature.rssi.leds=1
feature.ext_reset=0

mpannen1979, 2019-04-08:
Here is a factory support file too.
[support-XW-788A2092BEFD.zip](https://github.com/aredn/aredn_ar71xx/files/3052337/support-XW-788A2092BEFD.zip)

ae6xe, 2019-04-08:
next step, capture a support data download from the AREDN Administration page, link at bottom.   For some reason, this Ubiquiti board ID # is not found or correctly being returned, to track down.

mpannen1979, 2019-04-08:
It seems to work fine in my test setup despite the weirdness.

[supportdata-KB9OIV-146-190-253-201904072105.zip](https://github.com/aredn/aredn_ar71xx/files/3052370/supportdata-KB9OIV-146-190-253-201904072105.zip)

mpannen1979, 2019-04-08:
[image]

dman776, 2019-04-08:
You may also want to attach the output of http://localnode/cgi-bin/sysinfo.json

[Capture]<https://user-images.githubusercontent.com/29083576/55694447-dcb0cd00-5979-11e9-8db6-befa59a87165.PNG>

mpannen1979, 2019-04-08:
sysinfo.json:

{
   "lon": "-88.015928",
   "sysinfo": {
     "uptime": "0 days, 0:13:52",
     "loads": [
       0.059999999999999998,
       0.050000000000000003,
       0.059999999999999998
     ]
   },
   "interfaces": [
     {
       "mac": "78:8A:20:93:BE:FD",
       "name": "eth0.2",
       "ip": "10.147.190.253"
     },
     {
       "mac": "78:8A:20:92:BE:FD",
       "name": "wlan0",
       "ip": "10.146.190.253"
     },
     {
       "mac": "78:8A:20:93:BE:FD",
       "name": "br-lan",
       "ip": "10.149.247.233"
     },
     {
       "name": "tunl0",
       "mac": "00:00:00:00"
     },
     {
       "name": "eth0",
       "mac": "78:8A:20:93:BE:FD"
     },
     {
       "mac": "78:8A:20:93:BE:FD",
       "name": "eth0.1",
       "ip": "172.16.2.140"
     }
   ],
   "api_version": "1.6",
   "meshrf": {
     "status": "on",
     "chanbw": "10",
     "ssid": "AREDN-10-v3",
     "channel": "184"
   },
   "tunnels": {
     "active_tunnel_count": "0",
     "tunnel_installed": false
   },
   "lat": "43.001678",
   "grid_square": "EN53xa",
   "node": "KB9OIV-146-190-253",
   "node_details": {
     "description": "ubnt-lbe-m5",
     "model": "Unknown Hardware ",
     "board_id": "Ubiquiti Litebeam M5",
     "firmware_mfg": "AREDN",
     "firmware_version": "3.19.3.0"
   }
 }

dman776, 2019-04-08:
Can you ssh into the device and send the output of this command:
cat /sys/devices/pci0000:00/0000:00:00.0/subsystem_device

mpannen1979, 2019-04-08:
My device sys tree stops at:
/sys/devices

There is no 'pci0000:00' sub part.

root@KB9OIV-146-190-253:/sys/devices# ls -la
drwxr-xr-x    6 root     root             0 Dec 31  1969 .
dr-xr-xr-x   11 root     root             0 Dec 31  1969 ..
drwxr-xr-x    3 root     root             0 Apr  7 21:31 gpiochip1
drwxr-xr-x   15 root     root             0 Dec 31  1969 platform
drwxr-xr-x    6 root     root             0 Dec 31  1969 system
drwxr-xr-x    8 root     root             0 Dec 31  1969 virtual
root@KB9OIV-146-190-253:/sys/devices# pwd
/sys/devices
root@KB9OIV-146-190-253:/sys/devices#

mpannen1979, 2019-04-08:
Though I am not sure this is a valid thing to do, this returns nothing either:

root@KB9OIV-146-190-253:/sys# find | grep subsystem_device
root@KB9OIV-146-190-253:/sys#

mpannen1979, 2019-04-08:
root@KB9OIV-146-190-253:/sys# /usr/local/bin/get_boardid
Ubiquiti Litebeam M5
root@KB9OIV-146-190-253:/sys#

ae6xe, 2019-04-08:
What does the output of this command give (careful, not a linefeed, may blend in with next prompt):

dd if=/dev/mtd7 bs=1 skip=12 count=2 2>/dev/null | hexdump -v -n 4 -e '1/1 "%02x"'
This is where ubnt captures the board ID from flash on many devices.

mpannen1979, 2019-04-08:
root@KB9OIV-146-190-253:~# dd if=/dev/mtd7 bs=1 skip=12 count=2 2>/dev/null | hexdump -v -n 4 -e '1/1 "%02x"'
e865root@KB9OIV-146-190-253:~#

mpannen1979, 2019-04-08:
BusyBox v1.28.4 () built-in shell (ash)

              _____  ______ _____  _   _
        /\   |  __ \|  ____|  __ \| \ | |TM
       /  \  | |__) | |__  | |  | |  \| |
      / /\ \ |  _  /|  __| | |  | | . ` |
     / ____ \| | \ \| |____| |__| | |\  |
    /_/    \_\_|  \_\______|_____/|_| \_|
    AMATEUR  RADIO EMERGENCY DATA NETWORK
-----------------------------------------------
 * 1 Battery          Connect all devices
 * 2 POE injectors    Upgrade firmware to AREDN
 * 3 cat5 cables      Setup with your callsign
 * 1 UBNT NanoStation Point the Antenna
 * 1 ipCam            Welcome to the Mesh!
-----------------------------------------------
  3.19.3.0, r7676-cddd7b4c77
 ----------------------------------------------

> root@KB9OIV-146-190-253:~#

 dd if=/dev/mtd7 bs=1 skip=12 count=2 2>/dev/null | hexdump -v -n 4 -e '1/1 "%02x"'
e865root@KB9OIV-146-190-253:~#

ae6xe, 2019-04-08:
bingo.    There is an error in /usr/local/bin/get_boardid script.    your edit works with same end results.  Just need a minor change to key off this board id #, to be consistent with other devices.   This will be in the nightly builds shortly.

mpannen1979, 2019-04-08:
Looking at this file, I am not sure how it changes the boardid to a name, but sounds great! lol

ae6xe, 2019-04-08:
It's this case statement  the 'dd' command is only getting run for devices with images  "*-m-xw".  the option to do the dd command needs the "lbe-m5" image name included in the test.   Possibly the approach you have taken will be unique and work for the long term.  but to guarantee, this dd output should be used.

 case $AR71XX_BOARD_NAME in
    *-m-xw)
      BOARDID=$(dd if=/dev/mtd7 bs=1 skip=12 count=2 2>/dev/null | hexdump -v -n 4 -e '1/1 "%02x"')
      BOARDID="0x${BOARDID}"
      ;;
    *)
      BOARDID=$AR71XX_MODEL
      ;;
  esac

mpannen1979, 2019-04-08:
Ahhh.  Perfect, thanks for helping figure it out!

ae6xe, 2019-04-13:
mpannen1979, can you sysupgrade your LBE to the image here and confirm the change is working as expected, before I commit the change to the nightly build?

https://drive.google.com/drive/folders/1YcmhVOqDd-hY69fe05oQts881-blPVv7?usp=sharing

mpannen1979, 2019-04-13:
The power selection is good.
No more unsupported message.

That did the trick I think.

node: KB9OIV-146-190-253
model: LiteBeam M5 

br-lan 78:8A:20:93:BE:FD
eth0   78:8A:20:93:BE:FD
eth0.1 78:8A:20:93:BE:FD
eth0.2 78:8A:20:93:BE:FD
tunl0  00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
wlan0  78:8A:20:92:BE:FD
wlan0-1 7A-8A-20-92-BE-FD-00-00-00-00-00-00-00-00-00-00

/proc/cpuinfo
system type		: Atheros AR9342 rev 2
machine			: Ubiquiti Litebeam M5
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 266.64
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='78:8a:20:92:be:fd'
hsmmmesh.settings.mac2='146.190.253'
hsmmmesh.settings.dtdmac='147.190.253'
hsmmmesh.settings.node='KB9OIV-146-190-253'
hsmmmesh.settings.config='mesh'
hsmmmesh.settings.nodeupgraded='0'

closed 2019-04-14 by ae6xe

nerdoftech, 2019-05-16:
I had the same problem. Changing the file `/usr/local/bin/get_boardid ` as per fix #398 worked for me.

dman776, 2019-05-16:
This has already been fixed in PR#401 as noted above.

PR #401 (2019-04-14): bugfix: ubnt lbe-m5 does not identify boardid
fixes #398
commit 9665a8ae (2019-04-14): bugfix: ubnt lbe-m5 does not identify boardid

fixes #398