aredn/aredn_ar71xx #418: Extra long node name on iNet GL USB150 with Nightly Builds 902<>914

gelmce, 2019-05-13:
After assigning a node name in setup,
my stored node name is also shown as a tactical name and
"dtdlink." is prefixed and (tun*-1) is postfixed.
On the mesh /cgi-bin/mesh, it has a double node name ala:
dtdlink.NC8Q-USB.local.mesh / NC8Q-USB.local.mesh (tun*-1) 100% 100% 56.8

May the 'dtdlink.' and tactical name and the '(tun*-1) be removed?

dman776, 2019-05-13:
can you attach a support data file from the node?   *you may need to rename it to .zip to attach it*

gelmce, 2019-05-13:
can you attach a support data file from the node? Yes!
To keep my displayed node name shorter, I renamed my node
nc8q-u / nc8q

[supportdata-NC8Q-u-201905131801.gz](https://github.com/aredn/aredn_ar71xx/files/3174443/supportdata-NC8Q-u-201905131801.gz)

gelmce, 2019-05-13:
This is what its mesh neighbors display in /cgi-bin/mesh:

dtdlink.NC8Q-u.local.mesh / NC8Q.local.mesh |   | 100% | 100% | 72.2 |   |  
-----
I changed node name to "NC8Q-USB150", there is still a tactical node name added:

dtdlink.NC8Q-USB150.local.mesh / NC8Q.local.mesh |   | 89% | 100% | 65.0

Here is the new support data file:
[supportdata-NC8Q-USB150-201905131824.gz](https://github.com/aredn/aredn_ar71xx/files/3174523/supportdata-NC8Q-USB150-201905131824.gz)

dman776, 2019-05-13:
Try this:

Power off the node for 10 mins.
Power up.

Is the nodename display “correct”?   If so, this is a known behavior when changing the node name.   It takes about 10 mins of it being offline for all the other mesh nodes to flush the name from their name caches.

gelmce, 2019-05-13:
On 5/13/19 3:49 PM, dman776 wrote:
> Try this:
>
> Power off the node for 10 mins.
> Power up.
>
> Is the nodename display “correct”? If so, this is a known behavior 
> when changing the node name. It takes about 10 mins of it being 
> offline for all the other mesh nodes to flush the name from their name 
> caches.
>
Hi, Darryl:
  Not the issue. :-| I've had this off overnight and the issue remains. 
I've dealt with the issue you mention.
In this issue I often

ssh -p2222 root@<node>
/etc/init.d/olsrd stop; sleep 4; reboot

This usually fixes the new-node-name/old-node-name issue.

I'll leave it unplugged while I do an errand.

l8r

Chuck

dman776, 2019-05-13:
Chuck, when you first loaded AREDN from OpenWRT, did you UNCHECK the "Keep Settings" box?
If not, you should fully boot the node (in AREDN), go to the Advanced Configuration page and click the "Reset to Firstboot" button.  After it reboots, setup the node again and see if this issue returns.

gelmce, 2019-05-14:
Resolved. 'Save Settings' was not unselected when AREDN firmware was FIRST loaded.
Unselecting 'Save Settings' and uploading AREDN firmware resolved this issue.
Then the GL USB150 device boots to 'Firstboot' at 192.168.1.1:8080.
Set new node name and password, click 'Save Changes' and device will reboot itself.

closed 2019-05-14 by gelmce