aredn/aredn_ar71xx #424: Feature Request: add support for Ubiquiti Unifi Mesh / Unifi Mesh Pro

jmspswny, 2019-05-26:
I've got a couple of Unifi Mesh / Mesh Pros sitting around that I'd love to be able to deploy with this firmware.  They're compelling outdoor installation options.  They're both designed for outdoor deployment, with 802.3af support, and a couple of antenna options.  They're both supported by the current ar71xx release OpenWRT (18.06.2) and additional platform details are available here: https://openwrt.org/toh/ubiquiti/unifiac

I'm happy to help out with with this, just need some direction.

ae6xe, 2019-05-27:
Your devices are circular shape?  From openwrt link:

"The devices released in Q4/2015 (and listed here) are generation 2 devices based on ar71xx platforms. The UniFi AP AC Pro from generation 1 (based on some Broadcom platform) is not covered by this article. To easily identify gen 1 and gen 2: gen 2 devices have circle-shaped cases, while especially the generation 1 AP AC Pro has a square-shaped body."

I suspect these dual band devices have a 2Ghz 802.11n chip and the 5GHz is 802.11ac.  If so, then this would be similar to the hAP ac lite where we can only do an AREDN mesh on 2GHz.  The 5GHz becomes a mesh LAN AP in the shack, until some future time to figure out how to move the ac wireless driver into part97 channels. 

By digging around in AirOS or via command line, can you confirm the 2GHz wireless is on "wlan1" and is only 802.11n?

jmspswny, 2019-05-27:
my devices are the Atheros ar71xx devices, which come in a variety of shapes. More info on the AP Mesh here: https://store.ui.com/products/unifi-ac-mesh-ap
AC Mesh Pro here:
https://store.ui.com/products/unifi-ac-mesh-pro-ap

I've got these devices running OpenWRT 18.06.2, which has radio1 as the 2.4GHz 802.11bgn device.  radio0 is the 5GHz 802.11nac device.

ae6xe, 2019-05-27:
Ok, good.  These would behave the same as an hAP ac lite device and all the configuration and code exists in AREDN already to handle.   What does Openwrt have (the contents) of the files in, "/tmp/sysinfo/*"?  Just need this to set power settings in AREDN.

Are you running 18.06.2 on the unifi-ac-mesh-pro-ap?    On https://openwrt.org/toh/ubiquiti/unifiac, it shows only snapshot branch (I assume ar71xx snapshot)?  (It's not showing 18.06.2?)    The "unifi-ac-mesh-ap looks like 15 mins of edits and a image build, but not sure about the mesh-pro if it is not in 18.06.2...

jmspswny, 2019-05-27:
I've got a Mesh running 18.06.2 and a Mesh Pro running the snapshot.  I'll keep an eye on the Pro stuff to make it down to release and open another issue here when that happens.

In the mean time, the files from /tmp/sysinfo/ are attached:

[tmp-sysinfo contents.zip](https://github.com/aredn/aredn_ar71xx/files/3224962/tmp-sysinfo.contents.zip)

ae6xe, 2019-05-27:
Anyway to reload AirOS on this device and capture the /etc/board.info file?  (ssh in with ubnt@<IP address> and admin password or 'ubnt' if default).  

Unfortunately, the power offsets are not posted elsewhere, e.g. https://git.openwrt.org/?p=project/iwinfo.git;a=blob;f=hardware.txt .   the board.info file tells all.

jmspswny, 2019-05-27:
Here ya go.

[board.info.txt](https://github.com/aredn/aredn_ar71xx/files/3225031/board.info.txt)

ae6xe, 2019-05-27:
thanks.   openwrt indicates to use the "unifiac-mesh" image.   This is the 18.06.2 image you have installed, correct?   Internally, this must be the "unifiac-lite" target (which is what /tmp/sysinfo/board_name is showing.    Config options to build have both "unifiac-mesh" and "unifiac-lite" options that produce 2 different images,  but internally, there is only the "unifiac-lite" target.    Anyway, I'll build the unifiac-mesh build settings and image to try.

jmspswny, 2019-05-27:
I am using the 18.06.2 unifiac-mesh image.  The board_name file does indicate `unifiac-lite`.  As an added extra bit of confusion, the silk screen on the board calls it the pico.  So... yeah.  lots of names for a single piece of hardware :-/

ae6xe, 2019-05-27:
No 'factory' image is produced by the AREDN build or available on the openwrt downloads.  the tftp bootloader or AirOS UI upload process for this device accepts the sysupgrade.bin file from openwrt?    

Here's a test image:

 https://drive.google.com/file/d/1FQQiD0UJEBGrAQz52ht69WRTeOEzEBud/view?usp=sharing

standard disclaimer that your device could software brick :) , although haven't seen one yet that couldn't be recovered via serial console in a worse case.

jmspswny, 2019-05-28:
Installed the firmware.  Seeing some odd behavior with the mesh RF conjuration.  On initial config, it lists 5GHz channels, which it shouldn't.
[image]

After save, the overview shows it running on channel 153, which is again, from the 5GHz band.
[image]
So clearly things are still... fishy.

And then, when I login, the mesh rf configuration doesn't has a blank dropdown for channel selection:
[image]

Long story short - we're flashing and booting happily, but it seems we've crossed the streams in radio land.  

Let me know if there's more detail I can provide that would be helpful.

ae6xe, 2019-05-28:
capture the support download at bottom of Administration page and upload.

jmspswny, 2019-05-28:
here's that:

[supportdata-ww0o-2-201901300431.tar.gz](https://github.com/aredn/aredn_ar71xx/files/3225775/supportdata-ww0o-2-201901300431.tar.gz)

ae6xe, 2019-05-28:
The 2GHz radio0 is not detected for some reason and is a no-show.  If you turn off mesh RF, everything else might function as expected.    I'll need to dig though the build configs to compare with a 18.06.2 out-of-box to see if anything is different.   In parallel, you might reload 18.06.2 and confirm that both the 2Ghz and 5Ghz radios are working (or has the same issue).

jmspswny, 2019-05-28:
reflashed back to 18.06.2, where everything is working again.  Both radios show up and work as expected.

ae6xe, 2019-05-29:
can you post the output of dmesg on 18.06.2?   The build options for the 2GHz ath9k wireless driver are identical in AREDN and 18.06.2, so there is some subtlety somewhere else.

jmspswny, 2019-05-29:
Here's that output.

[dmesg-openwrt-18.06.2-unifi-mesh.txt](https://github.com/aredn/aredn_ar71xx/files/3230592/dmesg-openwrt-18.06.2-unifi-mesh.txt)

jmspswny, 2019-06-10:
[dmesg-openwrt-18.06.2-unifi-mesh-sample2.txt](https://github.com/aredn/aredn_ar71xx/files/3274118/dmesg-openwrt-18.06.2-unifi-mesh-sample2.txt)
Here's the dmesg from another unifi mesh, though this one isn't region locked to the US.

jmspswny, 2019-06-12:
and here's a non-region locked Unifi Mesh Pro dmesg from 18.06-snapshot 
[dmesg-openwrt-18.06-r10192-3b36aae-unifi-mesh-pro.txt](https://github.com/aredn/aredn_ar71xx/files/3279050/dmesg-openwrt-18.06-r10192-3b36aae-unifi-mesh-pro.txt)

patrickdickey52761, 2019-09-07:
> 
> 
> Anyway to reload AirOS on this device and capture the /etc/board.info file? (ssh in with ubnt@ and admin password or 'ubnt' if default).
> 
> Unfortunately, the power offsets are not posted elsewhere, e.g. https://git.openwrt.org/?p=project/iwinfo.git;a=blob;f=hardware.txt . the board.info file tells all.

I was just checking on this, as I found a deal for 9 of these. The datasheet (https://dl.ubnt.com/datasheets/unifi/UniFi_AC_APs_DS.pdf) shows the ac-lite as 20dBm and the AC-Pro as 22dBm for the maximum transmit power.