aredn/aredn_ar71xx #426: New device: WBS510 V2.0

xJMV, 2019-05-27:
Source of the request: https://www.arednmesh.org/content/support-tp-link-wbs210510

DMESG is not working... I don't remember any problems when I made the same process for the CPE510 V2/V3... ( See #288 ). From what I have read, CPE510 should be fairly the same device as the WBS.

Let me know if you need anything else!
 - VA2XJM

### DMESG
 Not working...
`BusyBox v1.01 (2017.09.08-04:27+0000) Built-in shell (ash)
Enter 'help' for a list of built-in commands.

-> help

Built-in commands:
        . : alias bg break cd chdir continue eval exec exit export false
        fg hash help jobs kill let local pwd read readonly return set
        shift times trap true type ulimit umask unalias unset wait

-> dmesg
-sh: dmesg: not found
`

### CPUInfo

`# cat /proc/cpuinfo
system type             : Atheros AR934x
processor               : 0
cpu model               : MIPS 74Kc V4.12
BogoMIPS                : 279.55
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0000, 0x0ff8, 0x0ff8, 0x0ff8]
ASEs implemented        : mips16 dsp
shadow register sets    : 1
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available
`

### cat /proc/cmdline
`console=ttyS0,115200 root=31:04 rootfstype=squashfs init=/init mtdparts=ath-nor0:128k(u-boot),64k(pation-table),64k(product-info),1536k(kernel),6144k(rootfs),192k(config),64k(ART) mem=64M`

### cat /proc/meminfo
```
MemTotal:          62572 kB
MemFree:           27820 kB
Buffers:            2696 kB
Cached:             9908 kB
SwapCached:            0 kB
Active:            10776 kB
Inactive:          10112 kB
Active(anon):       8284 kB
Inactive(anon):        0 kB
Active(file):       2492 kB
Inactive(file):    10112 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          8304 kB
Mapped:             3208 kB
Slab:               9480 kB
SReclaimable:        432 kB
SUnreclaim:         9048 kB
PageTables:          228 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:       31284 kB
Committed_AS:      15952 kB
VmallocTotal:    1048404 kB
VmallocUsed:        2864 kB
VmallocChunk:    1037444 kB
```

### cat /proc/devices
```
Character devices:
  1 mem
  4 ttyS
  5 /dev/tty
  5 /dev/console
  5 /dev/ptmx
 10 misc
 77 ATH_GPIOC
 90 mtd
108 ppp
128 ptm
136 pts
238 ar7100_gpio_chrdev
239 ar7240_flash_chrdev

Block devices:
259 blkext
 31 mtdblock
```

### cat /proc/mtd
```
dev:    size   erasesize  name
mtd0: 00020000 00010000 "u-boot"
mtd1: 00010000 00010000 "pation-table"
mtd2: 00010000 00010000 "product-info"
mtd3: 00180000 00010000 "kernel"
mtd4: 00600000 00010000 "rootfs"
mtd5: 00030000 00010000 "config"
mtd6: 00010000 00010000 "ART"
```

### ifconfig -a
```
ath0      Link encap:Ethernet  HWaddr 50:C7:BF:58:84:E4
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4 errors:0 dropped:247 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:948 (948.0 B)

br0       Link encap:Ethernet  HWaddr 50:C7:BF:58:84:E4
          inet addr:192.168.0.254  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1033 errors:0 dropped:0 overruns:0 frame:0
          TX packets:883 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:123739 (120.8 KiB)  TX bytes:520871 (508.6 KiB)

eth0      Link encap:Ethernet  HWaddr 50:C7:BF:58:84:E4
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1060 errors:0 dropped:0 overruns:0 frame:0
          TX packets:883 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:900
          RX bytes:146023 (142.6 KiB)  TX bytes:525771 (513.4 KiB)

eth1      Link encap:Ethernet  HWaddr 50:C7:BF:58:84:E4
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:900
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:3788 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3788 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:253372 (247.4 KiB)  TX bytes:253372 (247.4 KiB)

wifi0     Link encap:UNSPEC  HWaddr 50-C7-BF-58-84-E4-00-46-00-00-00-00-00-00-00-00
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5 errors:0 dropped:0 overruns:0 frame:0
          TX packets:17304 errors:1 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:2 Memory:b8100000-b811ffff
```

### brctl show
```
bridge name     bridge id               STP enabled     interfaces
br0             8000.50c7bf5884e4       no              eth0
                                                        eth1
                                                        ath0
```

apcameron, 2019-11-18:
I have it working on the [ATH79](https://github.com/apcameron/openwrt/tree/wbs510v2-ath79) branch of OpenWrt and will submit a PR for it to OpenWrt once I have finished testing. 
Next I will work on the code for the ar71xx Branch.
Here is a copy for the dmesg from the device on Openwrt
[wbs510v2_dmesg.txt](https://github.com/aredn/aredn_ar71xx/files/3856382/wbs510v2_dmesg.txt)

apcameron, 2019-11-19:
I have created a [PR](https://github.com/openwrt/openwrt/pull/2560)  to get if Officially supported under the ath79 branch.
I am working on the code for the ar71xx branch so we can create a PR for Arednmesh

xJMV, 2019-11-19:
Lightning fast! Thanks for your work mate!

apcameron, 2019-11-23:
I have created PR[#466](https://github.com/aredn/aredn_ar71xx/pull/466) to have supported added

apcameron, 2019-11-24:
It is now available in the nightly builds.

xJMV, 2019-11-24:
Will test this week when I can and return here.

Thanks a lot @apcameron !

xJMV, 2020-02-17:
Finally had time to flash the 2 devices I have home and made tests with them during the weekend.

Everything works as supposed.

Very about the outcome :)

Thanks @apcameron for the work!

ae6xe, 2020-04-22:
fixed by https://github.com/aredn/aredn_ar71xx/commit/5eedf421da0b4dfa19e5ba5b7868ed577881a243

closed 2020-04-22 by ae6xe