aredn/aredn_ar71xx #427: Update TP-Link network switch config

Orv, 2019-05-27:
Update code for TP-Link products so the internal switch behaves the same as the Ubiquiti products, i.e., convert to dumb non-VLAN switch.   (appears not to have been done yet)

ae6xe, 2019-05-27:
The models in scope are the following:  CPE210v1.0, CPE210v1.1,  CPE510v1.0, and CPE510v1.1.  Other newer versions of the TPLink devices only have 1 physical port.  (These older models would not be received if purchased new today, rather only available on used market.)

Orv, 2019-05-27:
So - no update planned?  Or?

ae6xe, 2019-05-27:
In queue.  If all models were affected it would be higher in queue.

Orv, 2019-05-27:
Good to know - thanks.

AA7AU, 2019-06-05:
> The models in scope are the following: CPE210v1.0, CPE210v1.1, CPE510v1.0, and CPE510v1.1. 

Don't forget my lonely CPE220v3. Thanks for keeping this topic at least somewhat alive.

- Don - AA7AU

apcameron, 2020-04-27:
PR #523 should address this

Orv, 2020-04-27:
I'll test in the next day or so.  Thanks, guys.
Orv W6BI

apcameron, 2020-04-27:
OK, Please wait until the PR has been approved.

Orv, 2020-04-27:
Of course :-)

apcameron, 2020-04-29:
This should be ready for testing now in the nightly build.

ae6xe, 2020-04-29:
https://github.com/aredn/aredn_ar71xx/pull/523#issuecomment-621347799

closed 2020-04-29 by ae6xe