aredn/aredn_ar71xx #428: hAP dropping LAN packets if there is a DtD connection

k6ccc, 2019-06-04:
First a little background.  From my house there is not an AREDN path to anything.  My only connection from home is a tunnel to a node at my office which in turn has a great RF path to the local network.  At home I have a Mikrotik RB952Ui-5ac2nD (hAP ac Lite) that provides me local AREDN access to the network via the tunnel.  To keep the story short, I have discovered that if there is a LAN connection between port 5 of the hAP (the DtD port) and the switch, then the LAN connection is dropping A LOT of packets - about 40% of pings fail.  This is 100% repeatable.  Originally I determined what was going on using nightly build 942, but I updated the hAP to nightly build 960 to see if that helped.  There was no change.  Here is a description of my LAN and a link to a drawing extract:

hAP port 1 (WAN port) connected to Mikrotik CSS326 managed switch port 18 - configured for VLAN 1 untagged (AREDN WAN)
hAP port 2 (LAN port) connected to Mikrotik CSS326 managed switch port 20 - configured for VLAN 5 untagged (AREDN LAN)
hAP port 5 (DtD port) connected to Mikrotik CSS326 managed switch port 16 - configured for VLAN 2 tagged (AREDN DtD)
 - Note that the port 5 connection is a custom cable that only passes the data wires and NOT the POE wires between the hAP and the switch.

Family room PC 2nd LAN port (AREDN) connected to Mikrotik CSS326 managed switch port 10 - configured for VLAN 5 untagged (AREDN LAN)

VLANs 1 & 5 are also routed via a VLAN trunk on port 3 to the garage data cabinet into another CSS326 managed switch.  In the data cabinet, VLAN 1 connects to one of my routers to gain internet access, and VLAN 5 connects to a RasPi that is running MeshChat.

Port 22 of the Family room CSS326 switch is a trunk port for an additional AREDN node that uses a single port - although there is nothing connected to it.
Port 24 of the Family room CSS326 switch is a trunk port to a Part 15 NanoStation M5 that goes to a nearby radio site to provide internet to that site.

There are no active ports on VLAN 2 on the Family room switch other than the hAP.  As soon as there is an active LAN connection between the hAP port 5 and and the switch, the LAN port (hAP port 2) starts dropping pings.

For my tests, I was pinging from the Family room PC to the hAP on either the WiFi address (using the node name) or by pinging the LAN IP address.  I was also pinging the node at my office and observed dropped pings from there at the same time as dropped pings to the local node (as expected).

The last test was to connect an AREDN node (Mikrotik RBLHG-5nD) to port 22 so that there was something to provide a DtD connection to the hAP.  Again when I enabled port 16 (the DtD LAN) in the Family room switch, the LAN started dropping pings.

Here is the LAN drawing extract:
http://k6ccc.org/AREDN/AREDN-extract.png

I am also posting this to the AREDN forum.

ae6xe, 2019-06-04:
k6ccc,  shot in the dark...  Confirm there's not both a  SFP S1 fiber and port 3 copper cables going to the garage switch with possibility of circular paths? 

Could you show though to see-all of the css326 configuration?  Screen shots similar to https://wiki.mikrotik.com/wiki/SWOS/CSS326-VLAN-Example 

Not seeing any red flags jumping out so far.   It appears vlan 1 is used for some defaults.   Suggest using a different vlan than '1' on port 18 inside the CSS326, to avoid unintended settings.  On port 16, where untagged traffic is unexpected, set a default vlan ID to something that is only assigned to port 16 and has zero chance of traffic going anywhere else. 

Not seeing problems elsewhere, thus burden is to show the CSS326 is not the contributing factor, e.g. can we reproduce symptoms in any other environment, 2nd set of eyes on the CSS326 config, etc.

k6ccc, 2019-06-04:
Joe, Good observation on the two paths between the Family room and garage CSS switches - however the optical link has not been installed.  Eventually that will be LACP group - gotta bury about 30 feet of conduit to make that happen however...

As for VLAN 1, that sort of has to be VLAN 1 because any single port AREDN that will likely eventually be connected to port 22 uses VLAN 1 (no choice) for the WAN connection.  The only ports that have VLAN 1 is the untagged port 18 for the hAP, the tagged port 22 for a future single port AREDN node, and the port into the router (part of a VLAN trunk from the garage switch) - and of course the VLAN trunk between the switches.  Personally I wish AREDN was not using VLAN 1, but that boat sailed a long time ago...  The router is a Mikrotik RB750Gr3 running RouterOS version 6.44.1.  The switches are running SwitchOS vrsion 2.9

One test I could do is direct connect the hAP LAN connection to the Family room PC.  that would completely remove the switches from the AREDN LAN connection.  That would kill my MeshChat, but that's not a big deal for a test...

Here are screen captures of the Port, VLAN, and VLANs screens on both switches.  Tomorrow I can post a complete LAN drawing.  Although the AutoCAD drawing is updated, I don't have a current pdf of it.  I will create a current pdf at work tomorrow (don't have AutoCAD at home).  The filenames should be obvious.

http://k6ccc.org/AREDN/Family_room_switch_ports.png
http://k6ccc.org/AREDN/Family_room_switch_VLAN.png
http://k6ccc.org/AREDN/Family_room_switch_VLANs.png
http://k6ccc.org/AREDN/Garage_switch_ports.png
http://k6ccc.org/AREDN/Garage_switch_VLAN.png
http://k6ccc.org/AREDN/Garage_switch_VLANs.png

k6ccc, 2019-06-04:
Just did a five minute test wherein I plugged the hAP LAN connection directly to the family Room PC.  Never dropped a ping for five minutes.  Reconnected via the Family Room switch and within seconds I was getting dropouts.  Turned off port 16 (the DtD link) and the dropouts stopped.

So, anyone see something with my switch config?  Or suggestions of something to try / test?

ae6xe, 2019-06-04:
regarding vlan 1, one design option is to have all mesh nodes route through dtdlink to the hAP AC Lite to access the internet and not directly have vlan 1 tagged extended to all nodes.   Port 1 of the hAP AC lite could be connected directly to an internet network ,etc. vlan 100, 101, etc. (untagged).   This does not appear to the problem, however.  

Try removing vlan 5 and vlan 2 from all other ports except the family room PC and the haplite.  Also, disconnect the network cable on the family room PC to 192.168.101.x network.    Can you get to a point where the symptoms can not be reproduced -- take to the simplest possible config of these vlans?   If necessary, remove haplite port 1 WAN cable too.     

We're hoping to see the symptoms go away.  If so, then add in the definitions one at a time to the other ports, until the problem returns to isolate which port is the contributing factor.

Noting is jumping out in the config screen snapshots.

k6ccc, 2019-06-04:
Several good suggestions to try Joe.  I likely won't get to them until Wednesday as I need to head in the direction of bed before long, and I will be out for the evening Tuesday.

I had thought earlier about having only the hAP have an internet connection as well, so that I can re-configure.  I'm assuming that at some point I will have a node that points somewhere to get me to the mesh from here :)

k6ccc, 2019-06-06:
Joe, I just sent you some info via a direct E-Mail.

AJ6GZ, 2019-06-06:
Try setting port 16 VLAN mode to "strict"?

k6ccc, 2019-06-06:
AJ6GZ, that is a valid suggestion.  To be honest, the use of Any and Optional is the defaults and not changing them is essentially being lazy.  You can see a few ports were changed, but most were left as the defaults because "well, it works".   This may be a situation where it does not.

I played with it for a while last night and could not get a situation where I was not getting dropouts on the LAN connection between the family room PC and the hAP port 2 if the DtD port was enabled.  However based on some of the earlier testing, I am reasonably convinced that the root cause of this is something in the switch since the problem goes away if I run a direct connection between the family room PC and the hAP port 2.  I will play with a few more of the switch settings tonight or Friday.  I also think I will ultimately change my settings so that only the hAP will have a WAN connection which because the hAP outputs the WAN connection untagged, can use any VLAN in the switch.

BTW Joe, on the drawings I sent you last night, I realized that the legend has been screwed up at some point.  Not that it's overly critical, but blue lines are 10base-T, green lines are 100base-T, and red lines are 1000base-T.  Fat red lines are gigabit optical (and not there yet).
More later...

k6ccc, 2019-06-07:
Last night, I tightened up the disabled vs optional vs strict assignments on the ports on both the family room and garage switches.  In the process, I locked myself out of the garage switch.  I did manage to get back into it - and learned something about two of the switch settings in the process.  In any case, the problem persists.  More adjustments this evening or over the weekend...

k6ccc, 2019-06-21:
I have tried all kinds of setting changes without any success, but been too busy to document it all here....  I'll try to do that this weekend.

ae6xe, 2019-08-25:
k6ccc,  was there any conclusion on this?   Not sure what could be done in the firmware, and vlan configuration on nodes is working in numerous other environments.

ae6xe, 2020-08-06:
Not sure what action can be taken, will close this issue.   If any concern and if there is a use case others can reproduce, then please submit a fresh issue.

closed 2020-08-06 by ae6xe