aredn/aredn_ar71xx #429: Mikrotik hAP AC Lite POE passthrough advanced setting is confusing

KC6RKG, 2019-06-06:
The advanced setting for POE Passthrough (aredn.@poe[0].passthrough) is confusing because it does not conform to standard checkbox usage which states: 

> With a check box, users make a decision between two clearly opposite choices. The check box label indicates the selected state, whereas the meaning of the cleared state must be the unambiguous opposite of the selected state.

The initial setting is a checkbox that was not checked and the label says "OFF". That implies that you check the box to turn off POE passthrough, and that with it unchecked, POE is ON. After checking the box it then showed a check mark and the label says "ON". The label (or what appears to be the label) should not change to an opposite state depending on whether or not the checkbox is checked. After toggling the check It now is obvious that the label is being used to show that state of the checkbox, but that is what the check mark is for, not the label for the check. You could either remove the label, in which case the Config Setting name would substitute for it, or you could just leave the label as "ON" and the checked state would indicate whether or not it is on. This issue applies to USB passthrough, too, and maybe other settings, too.

ae6xe, 2019-06-06:
KC6RKG,  re-writing the UI is occurring over at https://github.com/aredn/aredn_ui .   It's unlikely anyone will spend time to improve the current UI over replacing it.    Let's just say that there are no fans of the current UI -- generating HTML code from perl code is not a very effective or efficient way to do any UI.     Anyone with a background in UI development, particular angular, please reach out.

k1ky, 2019-08-05:
While you are at it - the olsr reboot feature selection feature is also a bit confusing (although very useful) - some cleanup (with the rewrite) would be appreciated there as well.

PR #560 (2020-08-26): enhancement: change advancedconfig boolean control
change widget rendering of boolean values on advancedconfig page to be more clear
resolves #429
closed 2020-08-26 by dman776