aredn/aredn_ar71xx #431: Nightly Build 960

ai6bx, 2019-06-18:
Nightly build 960 seems to possibly have some OLSR issues as nodes names are not consistently resolving to IP addresses and some nodes intermittently drop out as neighbors. I have seen this on Rocket Xw Nanoststion, and PBE 400.

martinalcock, 2019-06-19:
I confirmed last night that there is definitely a bug in the 960 daily build with respect to DNS lookup. Last night I did a demo at a club and had my USB microuter and a LinkSys set up on the same frequency. I checked before my presentation and all was fine. About a 45 minutes later at the end of the presentation, I showed the mesh status of the microuter, and the Linksys was still there, but I could not access it. Clicking the link ended up in a timeout. I was not able to access by IP simply because I did not know the current IP of the other station, but when I have seen this before access by IP usually works. I also have the same load in my hAP lite, and am noticing some problems with it as well. I have moved my services to another node so that they remain online.
 
I rebooted the microuter and it meshed again, this time I was able to access it from the status page. It appears that the bug is that the DNS entry for a node is somehow being corrupted or lost.  It is there to begin with, but a process in the router seems to be removing or corrupting them. Keith mentioned a new feature that did something similar, I did not get all the details, but it appears to be the culprit.

I have also seen this with the same software load on my hAP lite. It appears to lose the ability to resolve DNS addresses and times out, but attempting an access by IP will work most of the time.

ae6xe, 2019-06-20:
To confirm it is not this priour issue ( https://github.com/aredn/aredn_ar71xx/issues/204 ), please do the the test-workaround to see if the symptoms go away.   If a device is showing as an IP address or not resolving on other devices, then go to that device and from the command line,  "/etc/init.d/olsrd restart".  Wait ~30 seconds and confirm if symptoms have gone away or are still there.  Repeat a couple times if necessary.

When a PBE device in involved, confirm it is not this previous issue ( https://github.com/aredn/aredn_ar71xx/issues/165 ) and the workaround instructions to ensure the PBE is not trying to use 1G rates on the ethernet ports.  

Linksys, as in bbhn 3.x.x firmware?  If so, It's a honey pot for OLSR to lock up on an AREDN network, particularly with map crawlers.   It has none of the patches applied and very old firmware.   We're you connected to a greater mesh network or an island of 2 or 3 nodes?    Need a support data download to see the details on the AREDN nodes.   Guessing in the dark on a bbhn linksys...

ai6bx, 2019-06-20:
Joe, I checked both and this does not appear to be relics of the former issues.

Keith

martinalcock, 2019-06-21:
Definately not #204 or #165. I have seen the same on my home network with one hAP and two microuters at 960 and the rest running AREDN software of an earlier generation, I do not run a LinkSys in that network as I replaced it with a microuter. Did not see the problem until 960 was installed. 
In the case of my demo, it was an isolated island of two nodes, one microuter running 960 and a LinkSys running HSMM. The LinkSys was there to show the evolution to where we are now and how it played a role, in addition some members of my network have then running to satisfy the 'no RF connection' bug, which still seems to be out there. All ran properly at the onset, names were exchanged and links from the mesh status page worked correctly. After about 45 minutes, attempts to link from the mesh status page timed out, but all was restarted by rebooting the microuter. Fortunately I recovered by demonstrating how nodes mesh at power up!

ai6bx, 2019-06-23:
Joe,

I just sent you an email with greater detail and access info to nodes where you can observe the issues.

ai6bx, 2019-06-23:
One other item to note, if others are starting to track this issue, is the inconsistency of OLSR entries in nodes that are DTD linked or at close RF links. I have observed variances in record entries of at many as 35 entries where previously these numbers were equal or would become equal with a couple refreshes to pull records. Since 960 this is has been a persistent observation.

k6ccc, 2019-07-20:
For what it's worth, I have been seeing this issue at times with one radio on a DtD link (Mikrotik RB952Ui-5ac2nD (hAP ac Lite) to Mikrotik RB-LHG-5nD), however after updating all of my nodes to nightly 1022, the problem has become worse.  Still only sometimes, but more often.  The connected node will only report an IP address, and nodes another hop away don't show it at all in a mesh status.  
For example, at home right now, I have my Mikrotik hAP DtD connected to a Mikrotik LHG-5HPnD-XL, which is in turn RF connected to a Mikrotik LHG-5nD (about 50 feet away).
The LHG-5HPnD-XL is only reporting an IP address for the LHG-5nD.
The hAP does not report the LHG-5nD at all in the mesh status.
My computer (connected to the hAP) can only reach the LHG-5nD by IP address - not by name.
I have also observed this with the Rocket M5 at work that is RF connected to Verdugo Peak which is a a Rocket M5 XW.  Both of those are running nightly 1022.

Ten minutes later after a reboot of the hAP and the 5HPnD-XL, same connections.
The hAP is reporting only an IP for the LHG-5HPnD-XL and not seeing the LHG-5nD at all.
The LHG-5HPnD-XL is showing an IP for the LHG-5nD, but a name for the hAP.
The LHG-5nD is showing names for everything here at the house correctly.
The computer can reach only the hAP by name - both LHG nodes by IP only.

Does this all help?

ae6xe, 2019-07-20:
To confirm the root cause, check on remote nodes in /tmp/run/hosts_olsr.   The hostname is not being propagated by OLSR and available to do a nslookup of the name--the hostname is missing in the hosts table on remote nodes.    I'll need to find the time to roll up the sleeves, turn debugging on within OLSR, and find the breakdown.

k6ccc, 2019-07-20:
I don't speak enough linux.  I was able to cat that file and pipe it to more, but that is too long to be practical. I'm sure there is a better way.  Can you give me a command?

ae6xe, 2019-07-20:
Here is an example to search for a string in this file ("-i" means case insensitive).  If the node was not propagating a hostname, 1 or more of these entries might be missing.

root@AE6XE-NSM3-QTH:~# grep -i ae6xe-hap /tmp/run/hosts_olsr 
10.8.252.81	dtdlink.AE6XE-haplite-qth.local.mesh	# 10.7.252.81
10.7.252.81	AE6XE-haplite-qth	# 10.7.252.81
10.8.252.81	mid1.AE6XE-haplite-qth	# 10.7.252.81 (mid #1)

k6ccc, 2019-07-20:
Thanks.  I fully admit that I know enough linux to be dangerous!

root@K6CCC-hAP-at-Home:~/run# grep -i k6ccc /tmp/run/hosts_olsr
10.32.147.197   K6CCC-hAP-at-Home       # myself
10.4.158.44     K6CCC-RasPi-3B  # myself
10.4.158.45     K6CCC-GXP1625   # myself
10.33.147.197   dtdlink.K6CCC-hAP-at-Home.local.mesh    # myself
10.210.187.59   dtdlink.K6CCC-RB-LHG-5HPnD-XL.local.mesh        # 10.209.187.59
10.32.72.9      K6CCC-hAP-Portable      # 10.32.72.9
10.39.246.188   dtdlink.K6CCC-ROCM5-GLENDALE.local.mesh # 10.38.246.188
10.38.246.188   K6CCC-ROCM5-GLENDALE    # 10.38.246.188
172.31.127.237  mid1.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #1)
10.39.246.188   mid2.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #2)
172.31.90.77    mid3.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #3)
10.38.245.90    K6CCC-2-RM5     # 10.38.245.90

Interestingly the node K6CCC-RB-LHG-5HPnD-XL is listed, but it shows only with the IP on the hAP.
What is missing is the the LHG-5nD that is RF connected to the HP version.

I ran the same command on the HP version:
root@K6CCC-RB-LHG-5HPnD-XL:~# grep -i k6ccc /tmp/run/hosts_olsr
10.210.187.59   dtdlink.K6CCC-RB-LHG-5HPnD-XL.local.mesh        # myself
10.33.147.197   dtdlink.K6CCC-hAP-at-Home.local.mesh    # 10.32.147.197
10.4.158.45     K6CCC-GXP1625   # 10.32.147.197
10.4.158.44     K6CCC-RasPi-3B  # 10.32.147.197
10.32.147.197   K6CCC-hAP-at-Home       # 10.32.147.197
172.31.90.90    mid1.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #1)
172.31.90.86    mid2.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #2)
172.31.127.253  mid3.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #3)
172.31.90.78    mid4.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #4)
10.33.147.197   mid5.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #5)
10.32.72.9      K6CCC-hAP-Portable      # 10.32.72.9
10.39.246.188   dtdlink.K6CCC-ROCM5-GLENDALE.local.mesh # 10.38.246.188
10.38.246.188   K6CCC-ROCM5-GLENDALE    # 10.38.246.188
172.31.90.77    mid1.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #1)
172.31.127.237  mid2.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #2)
10.39.246.188   mid3.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #3)
10.38.245.90    K6CCC-2-RM5     # 10.38.245.90

And from the end of the line - the non HP that is RF connected to the HP version:
root@K6CCC-RB-LHG-5nD:~# grep -i k6ccc /tmp/run/hosts_olsr
10.33.147.197   dtdlink.K6CCC-hAP-at-Home.local.mesh    # 10.32.147.197
10.4.158.45     K6CCC-GXP1625   # 10.32.147.197
10.4.158.44     K6CCC-RasPi-3B  # 10.32.147.197
10.32.147.197   K6CCC-hAP-at-Home       # 10.32.147.197
172.31.90.90    mid1.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #1)
172.31.90.86    mid2.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #2)
172.31.127.253  mid3.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #3)
172.31.90.78    mid4.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #4)
10.33.147.197   mid5.K6CCC-hAP-at-Home  # 10.32.147.197 (mid #5)
10.210.187.59   dtdlink.K6CCC-RB-LHG-5HPnD-XL.local.mesh        # 10.209.187.59
10.209.187.59   K6CCC-RB-LHG-5HPnD-XL   # 10.209.187.59
10.210.187.59   mid1.K6CCC-RB-LHG-5HPnD-XL      # 10.209.187.59 (mid #1)
10.32.72.9      K6CCC-hAP-Portable      # 10.32.72.9
10.39.246.188   dtdlink.K6CCC-ROCM5-GLENDALE.local.mesh # 10.38.246.188
10.38.246.188   K6CCC-ROCM5-GLENDALE    # 10.38.246.188
172.31.90.77    mid1.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #1)
172.31.127.237  mid2.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #2)
10.39.246.188   mid3.K6CCC-ROCM5-GLENDALE       # 10.38.246.188 (mid #3)
10.38.245.90    K6CCC-2-RM5     # 10.38.245.90

ae6xe, 2019-07-21:
Summary

On host:  K6CCC-hAP-at-Home
missing entry:  10.209.187.59 K6CCC-RB-LHG-5HPnD-XL.local.mesh
missing entry:  10.210.187.59 mid1.K6CCC-RB-LHG-5HPnD-XL
10.210.187.59 dtdlink.K6CCC-RB-LHG-5HPnD-XL.local.mesh 

On host: K6CCC-RB-LHG-5HPnD-XL
missing entry:  10.209.187.59 K6CCC-RB-LHG-5HPnD-XL.local.mesh
10.210.187.59 dtdlink.K6CCC-RB-LHG-5HPnD-XL.local.mesh
(not not expected to see: 10.210.187.59 mid1.K6CCC-RB-LHG-5HPnD-XL on self node)

On host:  K6CCC-RB-LHG-5nD:
10.209.187.59 K6CCC-RB-LHG-5HPnD-XL      <-  Probably still cached from a prior time when it was working
10.210.187.59 dtdlink.K6CCC-RB-LHG-5HPnD-XL.local.mesh
10.210.187.59 mid1.K6CCC-RB-LHG-5HPnD-XL <- Probably still cached

If you reboot K6CCC-RB_LHG-5nD, I'd expect it will be missing entries too--if all nodes still in same state.

OLSR is not propagating the hostname and an olsr restart on K6CCC-RB-LHG-5HPnD-XL, most of the time fixes this -- which suggests there is a timing problem. 

Joe AE6XE

k6ccc, 2019-07-21:
Has this been useful?
BTW, if you had trouble trying to follow the node arrangement description, here is a drawing.
[image]
If that does not work right, the drawing is also at:
http://k6ccc.org/AREDN/K6CCC_AREDN_2019-07-20.png

ae6xe, 2019-07-21:
I don't need further information at this point.   Just the time to setup a debug environment that can reproduce.

k6ccc, 2019-07-21:
Understand.  BTW, I realized that my drawing had an error.  The link above has a corrected version.  The error was the node name of the upper center block.

mpannen1979, 2019-10-30:
I am seeing the same issue with some hAP AC lite that I have in a test setup with easy access.  

So far, I have only seen this happen to a hAP.  The name/IP doesnt make it across the mesh in the status page, but if I browse to a dtd neighbor of the suspect hAP, I can see the IP address listed in 'Current Neighbors' and (?,wan)

I did restart olsrd and that fixed it.  When I flashed 10/30/19 firmware and rebooted both hAP, one of the hAP resorted back to this condition again.

I am pretty handy with linux
[image]

mpannen1979, 2019-10-30:
There are no entries in the /tmp/run/hosts_olsr on the suspect hAP for "myself" when this happens.