aredn/aredn_ar71xx #442: PPB Gb port and toughswitch issue

ai6bx, 2019-08-10:


ai6bx, 2019-08-10:
Trying to track down an issue with a DMR repeater disconnecting issue with Brandmeister I started looking at SNR and link quality. Both appeared fine so I started looking for packet loss issues. One end of this path has a PBE 400 m5 ISO connected to a ToughSwitch 5 where it makes a dtd link to a NS M2 XW serving as a WAN gateway. The other end is a Rocket M5 XW. The PBE is running NB 1000 with ethtools applied and the TS is locked at 100FD. Here is where I noticed substantial RX packet loss. Where all my other switches are clean. As a test, I changed the switch port to auto neg which instantly cleaned up the packet loss but began creating the imbalance between LQ and NLQ. Thinking this might be a switch issue, I removed to another switch where I have a similar set up with one PBE and several Rockets. I found the same situation happening. It looks like the gigabit port issue appeared corrected but was actually masked in one aspect and pushed down the line.

ae6xe, 2019-08-25:
reference this on https://github.com/aredn/aredn_ar71xx/issues/165 and close this. Consolidate on one issue.

closed 2019-08-25 by ae6xe