aredn/aredn_ar71xx #443: Xiaomi routers

ea2bb, 2019-08-12:
It is possible to consider using Xiaomi router with Aredn?. It is a very sold brand and there is no Rom to use AREDN in them. Tnx

dman776, 2019-08-12:
In doing a quick look...

only 6 Xiaomi devices are supported by OpenWrt (which we are based upon)
only 1 of those devices uses an Atheros chipset (which we need for Part 97 support)
none of these devices are external/outdoor rated (which is our primary focus) (although we do already have a fairly large number of indoor devices from Mikrotik, Ubiquiti, and GL-Inet)

What advantages do these Xiaomi devices have over our existing supported devices?

ae6xe, 2019-08-14:
Is there a specific model proposal?   options would include 802.11n atheros-qualcomm chip sets.    Difficult to take action otherwise.

ae6xe, 2019-08-25:
There are some models supported by OpenWRT:
https://openwrt.org/toh/xiaomi/start

A couple of hurdles.  1) unknown if the wireless driver can be put into part 97 channels; 2) indoor packaging.    

I'm going to close this.  If there is a specific model proposal, then let's open up another ticket.

closed 2019-08-25 by ae6xe