aredn/aredn_ar71xx #450: Node losing network on a regular basis (Rocket M5)

idecmesh, 2019-09-18:
For at least the past year or two, I am experiencing the below symptoms on a Ubiquity Rocket M5 with AREDN firmware. I went through the different firmwares over these years and the problem has always been there.

[image]

When there is no activity on the node other than a few pings, everything is mostly stable. But as soon as you put some activity (pulling a web page, making an FTP transfer, etc.) within a few seconds the node will lose the network for a few seconds, and come back. So a constant ping to the node's IP results in:

[image]

In the beginning I thought it was the node self-rebooting, but when connecting SSH, I saw that the uptime is untouched.

The problem can be replicated 90% of the time. Doesn't have to be a huge traffic: go to the configuration page of a remote IP phone, and after browsing through a few pages the node will disappear for 6-to-10 seconds. Some things will never work, such as a speedtest.net, and will crash within the first seconds. 10% of the time, the node won't stop for a whole 2~3 minutes despite activity.

If I have an SSH client logged into the node doing a "ping localhost", I won't see the results while the network to the node goes down, and once the node network is back up, it will either show everything that has been buffered (no ping missed), or it will show "ping: sendto: Operation not permitted":

[image]

A couple of things I have noticed overtime:

* Immediately after a node reboot, everything is stable for at least 30 seconds to 1 minute, complete tasks that would normally stop in the middle (such as speedtest.net)
* Prior to the node's network disappearing, the node response time is slower. For example, if I try to download a nightly built, the browser estimates it will take 20 seconds, and after 12 seconds the node network disappear (making the download timeout, and I delete the partial file). When then network comes back, I start the download again, and now it takes 3 seconds to download everything.

With such behavior, my node has been unusable (can't transfer files, can have a complete SIP conversation). Any help getting this fixed would be much appreciated !

Thanks !

dman776, 2019-09-18:
@ae6xe this sounds exactly like the power beam gigabit Ethernet port issue, doesn’t it?   I didn’t think Rocket LD had a gigabit port?

dman776, 2019-09-18:
By the way, @idecmesh, great report.   Very thorough and detailed.   Can you attach a support data file from this node?

idecmesh, 2019-09-19:
[supportdata-W6XXX-RM5-201909190151.zip](https://github.com/aredn/aredn_ar71xx/files/3630328/supportdata-W6XXX-RM5-201909190151.zip)

Thank your for your superfast response @dman776 ! I actually wish I could provide more test data, but not sure where I should probe and collect from. If you want to assign me specific tasks I'll be happy to research and hopefully benefit everyone from the findings.

Attached is the support data file. Let me know if there is anything else I can do.

Thanks again!

ae6xe, 2019-09-22:
There are errors in the system logs of the link going up and down, presumably when accessing.   This is the 100Mb chip (no Gb support):

[    1.435065] ag71xx ag71xx.0: connected to PHY at ag71xx-mdio.0:04 [uid=004dd072, driver=Atheros 8035 ethernet]

But seeing these errors:
[  142.575123] eth0: link down
[  142.578460] br-lan: port 1(eth0) entered disabled state
[  162.336335] eth0: link up (100Mbps/Full duplex)
[  162.340997] br-lan: port 1(eth0) entered blocking state
[  162.346379] br-lan: port 1(eth0) entered forwarding state
[  329.775114] eth0: link down
[  329.778449] br-lan: port 1(eth0) entered disabled state
[  349.536279] eth0: link up (100Mbps/Full duplex)
[  349.540974] br-lan: port 1(eth0) entered blocking state
[  349.546331] br-lan: port 1(eth0) entered forwarding state
[  885.135108] eth0: link down
[  885.138437] br-lan: port 1(eth0) entered disabled state
[  903.856252] eth0: link up (100Mbps/Full duplex)
[  903.860906] br-lan: port 1(eth0) entered blocking state
[  903.866249] br-lan: port 1(eth0) entered forwarding state
[ 1159.695107] eth0: link down
[ 1159.698450] br-lan: port 1(eth0) entered disabled state
[ 1178.416263] eth0: link up (100Mbps/Full duplex)
[ 1178.421183] br-lan: port 1(eth0) entered blocking state
[ 1178.426565] br-lan: port 1(eth0) entered forwarding state
[ 1210.655136] eth0: link down
[ 1210.658466] br-lan: port 1(eth0) entered disabled state
[ 1248.096288] eth0: link up (100Mbps/Full duplex)
[ 1248.100939] br-lan: port 1(eth0) entered blocking state
[ 1248.106285] br-lan: port 1(eth0) entered forwarding state
[ 2011.455099] eth0: link down
[ 2011.458429] br-lan: port 1(eth0) entered disabled state
[ 2031.216264] eth0: link up (100Mbps/Full duplex)
[ 2031.221117] br-lan: port 1(eth0) entered blocking state
[ 2031.226478] br-lan: port 1(eth0) entered forwarding state
[ 3875.135171] eth0: link down
[ 3875.138495] br-lan: port 1(eth0) entered disabled state
[ 3893.856261] eth0: link up (100Mbps/Full duplex)
[ 3893.860916] br-lan: port 1(eth0) entered blocking state
[ 3893.866286] br-lan: port 1(eth0) entered forwarding state
[ 6433.695124] eth0: link down
[ 6433.698455] br-lan: port 1(eth0) entered disabled state
[ 6474.256211] eth0: link up (100Mbps/Full duplex)
[ 6474.260869] br-lan: port 1(eth0) entered blocking state
[ 6474.266218] br-lan: port 1(eth0) entered forwarding state
[ 8138.335054] eth0: link down
[ 8138.338378] br-lan: port 1(eth0) entered disabled state
[ 8158.096223] eth0: link up (100Mbps/Full duplex)
[ 8158.100890] br-lan: port 1(eth0) entered blocking state
[ 8158.106232] br-lan: port 1(eth0) entered forwarding state
[15626.415047] eth0: link down
[15626.418383] br-lan: port 1(eth0) entered disabled state
[15645.136468] eth0: link up (100Mbps/Full duplex)
[15645.141324] br-lan: port 1(eth0) entered blocking state
[15645.146683] br-lan: port 1(eth0) entered forwarding state
[17346.574992] eth0: link down
[17346.578334] br-lan: port 1(eth0) entered disabled state
[17379.856187] eth0: link up (100Mbps/Full duplex)
[17379.860863] br-lan: port 1(eth0) entered blocking state
[17379.866208] br-lan: port 1(eth0) entered forwarding state
[20203.454974] eth0: link down
[20203.458410] br-lan: port 1(eth0) entered disabled state
[20223.216126] eth0: link up (100Mbps/Full duplex)
[20223.220876] br-lan: port 1(eth0) entered blocking state
[20223.226259] br-lan: port 1(eth0) entered forwarding state
[20531.054982] eth0: link down
[20531.058570] br-lan: port 1(eth0) entered disabled state
[20549.776809] eth0: link up (100Mbps/Full duplex)
[20549.781921] br-lan: port 1(eth0) entered blocking state
[20549.787323] br-lan: port 1(eth0) entered forwarding state
[... 59 lines cut ...]
[37241.145994] br-lan: port 1(eth0) entered forwarding state
[38324.894848] eth0: link down
[38324.898186] br-lan: port 1(eth0) entered disabled state
[38350.895972] eth0: link up (100Mbps/Full duplex)
[38350.900855] br-lan: port 1(eth0) entered blocking state
[38350.906216] br-lan: port 1(eth0) entered forwarding state
[42857.214798] eth0: link down
[42857.218133] br-lan: port 1(eth0) entered disabled state
[42876.975937] eth0: link up (100Mbps/Full duplex)
[42876.980773] br-lan: port 1(eth0) entered blocking state
[42876.986129] br-lan: port 1(eth0) entered forwarding state
[44033.534955] eth0: link down
[44033.538278] br-lan: port 1(eth0) entered disabled state
[44053.295991] eth0: link up (100Mbps/Full duplex)
[44053.300705] br-lan: port 1(eth0) entered blocking state
[44053.306056] br-lan: port 1(eth0) entered forwarding state
[44124.094786] eth0: link down
[44124.098123] br-lan: port 1(eth0) entered disabled state
[44143.856014] eth0: link up (100Mbps/Full duplex)
[44143.860672] br-lan: port 1(eth0) entered blocking state
[44143.866020] br-lan: port 1(eth0) entered forwarding state

See if we can isolate the issue.  Are you able to plug in a laptop directly to the Rocket, with no switch in the middle?   Can you still reproduce these symptoms?

Joe AE6XE

idecmesh, 2019-09-23:
I can absolutely do the test @ae6xe - Can you please let me know how to look at this error log ? Thanks!

ae6xe, 2019-09-24:
Telnet or ssh into the node, and type the command "dmesg".

idecmesh, 2019-09-27:
It is now connected directly. Waiting to see results :-)

idecmesh, 2019-09-27:
Upon connecting directly behind the node, the **very same** symptoms were happening.

Then @dman776 comments got me thinking about the port, and I remember seeing problems like this in the past with auto-negotiation (especially on ISP-managed equipment). So I connected the node back on the switch, and went in the switch configuration to remove auto-negotiation and FORCE 100 Meg / Full-Duplex. Since then, there has been NEARLY NONE of the symptoms are happening. It's only been a couple of hours thus far, I will come back for updates.

Thanks for the team efforts, much appreciated.

ae6xe, 2019-10-06:
update:   locked switch to 100Mbps and problem was solved.   Since we don't see symptoms across many devices and switches, will close this as an isolated case -- likely unique to the ethernet implementation of this particular switch and the Rocket.

closed 2019-10-06 by ae6xe