aredn/aredn_ar71xx #452: Enhancement request: Mesh repository & Maps

xJMV, 2019-09-22:
To make upgrade process of AREDN simplier and less bandwidth hungry, we have decided to run our own AREDN mirror that is accessible over the mesh. This mirror looks monthly for new version of AREDN (From the stable repo only, not the nightly).

So we changed the paths in the advance configuration section to reflect our mesh server. The problem is that nodes return «Error: no route to the Internet» when trying to refresh the firmware list. The map on the «Basic Setup» doesn't appear. This behavior is observed on nodes without internet access.

If the internet is available, the data (map tiles, leaflet.js, available firmware version...) are fetched from our mesh server as we wished.

It seems obvious to me that there is a «safety» switch in the code to ensure no hanging requests are made. This safety mechanism should detect mesh link and proceed from the mesh even if internet is not available.

Screenshot joined. All devices except the GL-USB150 runs 3.19.3.0.

[image]
[image]
[image]

r1de, 2019-10-17:
Another thing on this is even if that "internet check" is totally disabled (it's not hard to comment out a couple of lines in the admin file), the `firmwarepath` setting from the advanced config page is totally ignored.  
The uci setting *does* indeed get changed as can be verified by:  
```
root@KG6WXC-VC-RF-5G:~# uci show aredn.@downloads[0].firmwarepath
aredn.cfg018df0.firmwarepath='http://kg6wxc-srv.local.mesh/aredn_nightly/1158'
```

opkg doesn't seem to care much though. It's looking for `/etc/opkg/distfeeds.conf` which is still pointed at downloads.arednmesh.org.  

(If I build my own firmware and change the value in config.mk, it just works, and none of this needs to be done btw.)

r1de, 2020-02-14:
I am going to work on this.

Should the "internet check" be a user selectable thing?
Should it be more automatic, like `if ($url contains "local.mesh") {`?

Also, if we're going to do this for the node's map and the firmware download, we might as well do it to the package downloads too.
Should we put the package URL's into the adv config page as well?
They will probably need to be, not everyone is going to have the same directory structures.
There are 7 different URL's that will need to be updated that are in the `/etc/opkg/distfeeds.conf` file.

xJMV, 2020-02-14:
@r1de I see it more automatic.

:+1 for the packages. It should also be in the adv config just like the other settings. 

For the directory structure, it should be easy to provide a one liner to clone the repos so structures are kept. This is how I do it right now with cronjobs (I do not have packages, but will do soon):
`# Get AREDN firmware every month & remove indexes files.
30      0       1         *     *       cd /media/sf_mirror/AREDN && wget -nd -r -c -l 1 http://downloads.arednmesh.org/firmware/ubnt && rm index.html*
# Remove files older than 365days (remove past release)
0       0       1         *     *       find /media/sf_mirror/AREDN* -mtime +365 -exec rm {} \;`

r1de, 2020-02-15:
I think I see it as more automatic too...
We'll see what the bosses say though. :)

I just started looking at this and one thing I see so far, is how to save the "default" settings for the package repo's?
There are several URL's now, based on chipset (ar71xx or ath79) and the manufacturer like "mikrotik" or "ubnt".
I cannot just hardcode the defaults in for this.
For instance, on my MikroTik HaP lite, I end up with a default URL (for the "core" packages) of:
`http://downloads.arednmesh.org/snapshots/trunk/targets/ar71xx/mikrotik/packages`
That URL won't work for UBNT devices.

Going to have to find an elegant way of saving the original info in the `/etc/opkg/distfeeds.conf`

r1de, 2020-02-15:
@xJMV 
try this in your cronjob:
```
wget --recursive -nH --cut-dirs=2 --no-parent --reject="index.html*" http://downloads.arednmesh.org/snapshots/trunk/
```
:)

xJMV, 2020-02-22:
@r1de was playing with Apache2 when I realized we can use a similar way to activate/deactivate configs...

Basically, Apache2 use 2 folder, the first one (sites-available) is where the configs are done. When you request to enable a config, A2 then create a symlink in the second folder (sites-enabled), where it reads the active configuration to include.

So it would mean the on boot/value modification, the node check the URL and symlink OR include whatever file is needed.

r1de, 2020-02-24:
it's not quite that simple...
Hopefully this will work: #486 
https://github.com/aredn/aredn_ar71xx/pull/486/files

r1de, 2020-02-25:
Since I was playing with this, here's a one liner to pull the entire "stable" directories too.

`wget -l 0 --recursive -nH --no-parent -I releases,firmware --reject="index.html*" http://downloads.arednmesh.org`

put that into your cron and smoke it! :)

PR #490 (2020-04-03): enhancement: ability to change and revert firmware and package downlo…
…ad paths

replaces #486
fixes #452
commit 83675fbd (2020-04-03): enhancement: ability to change and revert firmware and package download paths (#490)

replaces #486
fixes #452
closed 2020-04-03 by dman776