aredn/aredn_ar71xx #455: Package management - Wrong Signature File

idecmesh, 2019-09-29:
It seems there is a problem with packages signature file. The issue can be replicated 100% of the time with the below steps. Tried with few nodes on version: **1101-ad0caaf** and hardware type: ubnt (rocket-m). The consequence of this is that you can't install any package on the node.

========================================

Near "Download Package" - Drop down is empty -  Clicking the [Refresh] button.
Waiting and getting the following messages:

========================================

`Downloading
http://downloads.arednmesh.org/snapshots/trunk/targets/ar71xx/generic/packages/P
ackages.gz
Updated list of available packages in /var/opkg-lists/aredn_core
Downloading
http://downloads.arednmesh.org/snapshots/trunk/targets/ar71xx/generic/packages/P
ackages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/base/Packages.
gz
Updated list of available packages in /var/opkg-lists/aredn_base
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/base/Packages.
sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/arednpackages/
Packages.gz
Updated list of available packages in /var/opkg-lists/aredn_arednpackages
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/arednpackages/
Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/luci/Packages.
gz
Updated list of available packages in /var/opkg-lists/aredn_luci
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/luci/Packages.
sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/packages/Packa
ges.gz
Updated list of available packages in /var/opkg-lists/aredn_packages
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/packages/Packa
ges.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/routing/Packag
es.gz
Updated list of available packages in /var/opkg-lists/aredn_routing
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/routing/Packag
es.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/telephony/Pack
ages.gz
Updated list of available packages in /var/opkg-lists/aredn_telephony
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/telephony/Pack
ages.sig
Signature check failed.
Remove wrong Signature file.`

idecmesh, 2019-09-29:
Tried to reboot but didn't help

dman776, 2019-09-29:
You need to load the latest beta firmware.   The packages are tied to specific versions

Thanks,
Darryl

idecmesh, 2019-09-29:
Sorry about that, I didn't know the "Refresh" button download packages for the latest version rather than it's "current version". I just upgraded the node to the latest nightly build and was able to refresh/download packages. Thank you!!

closed 2019-09-29 by idecmesh