aredn/aredn_ar71xx #456: Add support for Ubiquiti Nanobeam M5 19 (Atheros AR934x)

radar27, 2019-10-02:
There appears to be at least one additional hardware revision for the Ubiquiti Nanobeam M5 19.  In purchasing two lots on the used market, a batch with the System type: Atheros AR934x exhibited inconsistent behavior on the wired Ethernet port using both the current stable (3.19.3.0 / ubnt-loco-m-xw) as well as snapshot as of 10/1/19 (aredn-1138-6e19994-ubnt-loco-m-xw-factory.bin)

The wired network connectivity seems to be hit or miss for the untagged user traffic and seemingly unidirectional for VLAN2/dtdlink traffic.  Directly wired to test computer sporadically assigns an IP address with DHCP enabled.  When paired with a Ubiquiti Rocket M2 Ti through an unmanaged Ubiquiti Nanoswitch, the dtd connection displays LQ of 100% but NLQ of 0%, as if the Ethernet connection is somehow functioning in one direction.  This system briefly began passing dtd traffic after 36+ hours of being powered up at an unattended location.  Once the Nanobeam M5 19 was reset after trying a second Nanoswitch, the original behavior of "one way" dtd traffic resumed.

The wireless side seems to work just fine, held a fairly solid link 19 miles to another Nanobeam M5 19 with the "Atheros AR9342 rev 2" System type and suggested ubnt-loco-m-xw stable image.

This appears to have the same processor type as the TP-Link CPE610 as noted in issue [#219](https://github.com/aredn/aredn_ar71xx/issues/219).  This may be similar to the November 2017 test date issue on the Nanostation Loco M5.  I did attempt loading the aredn-3.19.3.0-ubnt-rocket-m-xw-factory.bin for the after November 2017 issue as well as the aredn-3.19.3.0-ubnt-nano-m-xw-factory.bin, just to test out all available XW firmware.  Unfortunately no other firmware seemed to provide usable wired ethernet, so I am hoping adding support for this version is possible.

Here are system details taken from AirOS XW 6.2.0

**cat /proc/cpuinfo**
system type             : Atheros AR934x
processor               : 0
cpu model               : MIPS 74Kc V4.12
BogoMIPS                : 266.24
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0000, 0x0530, 0x0788, 0x01b8]
ASEs implemented        : mips16 dsp
shadow register sets    : 1
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

**dmesg**
[    0.000000] Linux version 2.6.32.71 (buildd@builder) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.01 r18801) ) #1 Wed Jul 3 11:18:17 EEST 2019
[    0.000000] flash_size passed from bootloader = 8
[    0.000000] CPU revision is: 0001974c (MIPS 74Kc)
[    0.000000] ath_sys_frequency: cpu srif ddr srif cpu 535 ddr 400 ahb 200
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   Normal   0x00000000 -> 0x00004000
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x00004000
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 8028a470, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16256 pages, LIFO batch:3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: console=tty0 root=31:03 rootfstype=squashfs init=/init mtdparts=ath-nor0:256k(u-boot),64k(u-boot-env),1024k(kernel),6528k(rootfs),256k(cfg),64k(EEPROM) mem=64M panic=3 quiet
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 61936k/65536k available (2081k kernel code, 3532k reserved, 516k data, 132k init, 0k highmem)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:128
[    0.000000] CPU0 IRQ stack at 0x83810000 - 0x83814000
[    0.000000] plat_time_init: plat time init done
[    0.000000] Console: colour dummy device 80x25
[    0.000000] console [tty0] enabled
[    0.000000] Calibrating delay loop... 266.24 BogoMIPS (lpj=133120)
[    0.022000] Mount-cache hash table entries: 512
[    0.024000] NET: Registered protocol family 16
[    0.032000] bio: create slab <bio-0> at 0
[    0.034000] Switching to clocksource MIPS
[    0.036000] NET: Registered protocol family 2
[    0.037000] IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.038000] TCP established hash table entries: 2048 (order: 2, 16384 bytes)
[... 166 lines cut ...]
radio.1.ccode_fixed=0
radio.1.ccode_locked=2
radio.1.cap.station_only=1;
radio.1.cap.ap_only=1;
radio.1.subsystemid=0xe825
radio.1.subvendorid=0x0777
radio.1.txpower.max=26
radio.1.txpower.min=-4
radio.1.txpower.offset=4
radio.1.ieee_modes=5
radio.1.antennas=1
radio.1.def_antenna=4
radio.1.antenna.1.id=4
radio.1.antenna.1.name=Built in
radio.1.antenna.1.gain=19
radio.1.antenna.1.builtin=1
radio.1.antenna.1.chains=2
radio.1.antenna.1.chain.1.id=0
radio.1.antenna.1.chain.1.name=Horizontal
radio.1.antenna.1.chain.2.id=1
radio.1.antenna.1.chain.2.name=Vertical
radio.1.distance_limit_km=0
radio.1.chanbw="5,8,10,20,30,40"
feature.rssi.leds=4
feature.ext_reset=0

ae6xe, 2019-10-02:
Can you load AREDN and work to access over the network port, then capture the support data download file (over RF if necessary through another node)?    The symptoms on the network port, since it is intermittent, suggests there might be a hardware issue.    There may be messages in the system logs of the network interface to give more information, I'd be looking for the interface going down/up repeatedly.  Generally, if there is a new hardware rev, then it ether works or it doesn't.

radar27, 2019-10-02:
Thanks for the prompt reply.  I too had originally thought it was hardware, I had a second Nanobeam M5 19 at the same location targeting a secondary 5 GHz link which was exhibiting similar behavior.  The interconnecting cables were tested before and after and showed continuity/order between all 8 pins and shield.  In the troubleshooting process, we replaced both the Nanoswitch and new tested cables and the behaviors persisted.  Attached is the support bundle for the AR934x in question (the .tgz was renamed to zip to allow upload).  

[supportdata-NOCALL-201901301222.zip](https://github.com/aredn/aredn_ar71xx/files/3679329/supportdata-NOCALL-201901301222.zip)

One other piece to this puzzle is that I reflashed this unit with the native AirOS 6.2.0 XW, connected to a local part 15 AP and was able to push approximately 35 megabit to an Internet speed test site through it in standard bridging mode (Laptop -> Nanobeam M5 19 -> Bullet M5 -> Firewall -> Internet).  This led me to believe the unit hardware was operating correctly.  I also tried moving the unit around from being directly into a laptop (via PoE injector) and also through an 10/100 unmanaged switch and discovered that changing between connections was resulting in intermittent connectivity even with AirOS.  

Lastly, in my subsequent testing, I also discovered that AirOS was reporting a different system type than the AREDN firmware on the same unit. 

XW.v6.2.0# cat /proc/cpuinfo
system type             : Atheros AR934x
processor               : 0
cpu model               : MIPS 74Kc V4.12
BogoMIPS                : 266.24

root@NOCALL:~# cat /proc/cpuinfo
system type             : Atheros AR9342 rev 2
machine                 : Ubiquiti Loco M XW
processor               : 0
cpu model               : MIPS 74Kc V4.12
BogoMIPS                : 266.64

Given all this, I think my correlation between the reported system type is not likely at fault here.  I am going to close this off and do more testing with the other units and combinations of switches/devices.  I apologize for misidentifying the issue, I now agree that this is most likely faulty hardware, though it's behavior has left me puzzled.  If I do uncover something consistent that can be reproduced on multiple units I'll be sure to check back in.  Thanks for all your excellent work!

closed 2019-10-02 by radar27

ampledata, 2020-01-17:
@radar27 I'm committing the royal sin of commenting on a closed issue, but I wanted to let you know that in the lot of NanoBeam M5-19s I've received, I've had at least 2 units exhibit the behavior you've described (I have several more left to test...)