aredn/aredn_ar71xx #462: Remove 'ether' option in /etc/config/olsrd to improve DtD link robustness

Orv, 2019-11-05:
In Ventura County we're able to locate radios on several elevated sites on private property in exchange for providing Internet access to the families living on these remote properties.  So those links have to be Part 15 radio links because they're carrying traffic that might violate Part 97 regulations.  We do partition VLAN 2 off on these trunks for management purposes.   

However DtD links default to a wired ('ether') status, and as such Joe has explained to me that the code is designed to drop out at an LQ lower than about 90 percent.  Frequently a Part 15 link can have enough interference enough to cause the LQ to drop below 90.   Additionally, a link carrying heavy but normal traffic can have its LQ drop when there's nothing wrong.  This premature disabling of a link causes unwanted dropouts, or even route flapping if there's an alternate route to the destination.

What we do to work around this is to remove the assumption in the AREDN config that every DtD link is a wired connection.  We do that by commenting out the line

option Mode 'ether'

in /etc/config/olsrd

Like this:

config Interface
        list interface 'dtdlink'
#        option Mode 'ether'

It's a hack, but it works.  The problem is, it's not persistent.  That is, every time you upgrade the software in a node, it reverts to default and has to be commented out again.

We in this area have concluded that the 'ether' option should just be removed completely.  If a DtD link (regardless if it's wired or wireless)  is having a problem and drops below 90 it should be allowed to keep passing traffic.  After all, we don't drop out RF links because they drop below 90.

k1ky, 2019-11-05:
Sounds like a good idea.  We also have VLAN2 traffic connecting nodes on a foreign Microwave network.  We are also going to be utilizing more RF links (non-mesh) as point-to-point connections over some longer haul paths in order to maximize throughput for the data.  Is there anything that can be done to enhance the Data throughput over our MESH links?

dman776, 2019-11-05:
This is a duplicate of https://github.com/aredn/aredn_ar71xx/issues/67

dman776, 2019-11-05:
Additionally, I think this would be a protocol breaking change.   DTDLinks across the mesh are expected to be highly reliable, WIRED connections.

closed 2019-11-05 by dman776

Orv, 2019-11-05:
Honestly, that's an unrealistic expectation (borne out by several years 
experience). This is an opportunity to tune the behavior of the code to 
real-world environments.

ae6xe, 2019-11-06:
This request would change the ETX calculations, and create other problems.   A DtDLink for a wired connection would change from a cost of 0.1 to 1.0.   This would have a negative impact on routing decisions to not distinguish or give any added bias to a wired link over an RF link.   As an example, some sites may have both RF And DtDlink paths between 2 nodes.  We'd want to avoid this to begin with, but in some cases it exists.   This change would no longer ensure the DtDlink hard wired connection  would be selected and increase symptoms of the route flapping weakness in OLSR.   We may create more problems then are being solved.

Orv, 2019-11-06:
Joe, I'm going to disagree.  While it would change the ETX calculations, that's legitimate and appropriate.  A DtD link over wireless should NOT be considered as a cost of 0.1.    

As you pointed out, paths with parallel RF and DtD links are extremely rare.  This fix would give us some robustness and counter route flapping by hanging on to a DtD route longer before giving up on it.

ae6xe, 2019-11-06:
A DtD link over a cat5 should be considered 0.1, that is also legitimate and appropriate.  We're trying to meet both requirements and get the best of both worlds.  Degrading one in favor of the other compromises.  We can have an implementation that meets the needs of both, with out compromising.  There's also discussion for a 3rd, to support NPR usage.

k1ky, 2019-11-06:
Actually, we have several situations where we have DTD links and RF links between some of our nodes. Both are RF - one dedicated 3rd party Microwave network plus our over the air MESH links. This is somewhat of a "test" situation to prove a concept.  Been running this for over a year now without any noticeable problems.

blm2008, 2019-11-06:
It sounds like Orv and others have come up with a workable solution.  Is the only concern around *always* having the dtd link add 1.0  instead of 0.1 to the ETX calculation if the olsr 'ether' parameter were permanently removed?  Could this be another 'advanced configuration' variable with a description and appropriate warning for users?  I'm glad that there is finally a solution to this need (Request #67 has been open for over a year) even if it requires manual changes to the olsr config.