aredn/aredn_ar71xx #465: New model - Mikrotik RB911G-2HPnD

kc8ufv, 2019-11-18:
This is a request for a new model to be supported.  Below is a copy of the unsupported model info. This board is supposed to be the same board as in the mANTBox 2 12s. I am working with the raw board.

 node: KC8UFV-RB911G-2HPnD
model: Unknown Hardware 

!!!! UNSUPPORTED DEVICE !!!!
boardid: Mikrotik RouterBOARD 911G-2HPnD
Device has not been tested. Please file a ticket with your experiences.

br-lan B8:69:F4:A7:31:88
eth0   B8:69:F4:A7:31:88
eth0.1 B8:69:F4:A7:31:88
eth0.2 B8:69:F4:A7:31:88
tunl0  00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
wlan0  B8:69:F4:A6:31:88
wlan0-1 BA-69-F4-A6-31-88-00-00-00-00-00-00-00-00-00-00

/proc/cpuinfo
system type		: Atheros AR9342 rev 3
machine			: Mikrotik RouterBOARD 911G-2HPnD
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 299.82
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='b8:69:f4:a6:31:88'
hsmmmesh.settings.mac2='166.49.136'
hsmmmesh.settings.dtdmac='166.49.135'
hsmmmesh.settings.node='KC8UFV-RB911G-2HPnD'
hsmmmesh.settings.config='mesh'
Part of the AREDN™ Project. For more details please see here

ae6xe, 2019-11-18:
kc8ufv, can you please capture the support download data, link at bottom of admin page, and attach back here.    The change for the mAntBox will be in the images produced by tonight's build, available in the AM.   Confirm status with these images.

kc8ufv, 2019-11-19:
Got it.

kc8ufv, 2019-11-19:
Looks good. Got 2 attached. the K8MWS-mANTBox one is from Mark (on the
forums under his old call KE8LUN), which is a full mANTBox, my supportdata
is from a bare board.

PR #528 (2020-05-11): ar71xx: add early support for Mikrotik r2 hardware
Apply openwrt PR 3006
https://github.com/openwrt/openwrt/pull/3006
author: Thibaut VARÈNE <hacks@slashdirt.org>

fixes: #465
early patch to AREDN will be removed and inherited from openwrt in
future 19.07 point or later release.  This patch enables use
of newer BaseBox, AntBox, and possibly other updated devices with ath9k
calibration data encoded with a different technique.
closed 2020-05-11 by ae6xe