aredn/aredn_ar71xx #471: NanoBeam M5-19 XW not seeing MAC(s)

ampledata, 2020-01-17:
Related to #456. I've had at least two NanoBeam M5-19's fall into this state. I can access the device using "MeshNode5G" SSID, but cannot access eth or change WiFi channel. Support file attached.

Support file: https://www.dropbox.com/s/sldq7loxoz9ztfo/supportdata-NOCALL-201901301241.tgz?dl=0

Screenshot: https://www.dropbox.com/s/vu4r7xm8wxwhhim/Screenshot%202020-01-17%2010.46.11.png?dl=0

ae6xe, 2020-01-17:
linux doesn't find the expected physical Ethernet chip:

[    0.692181] libphy: Fixed MDIO Bus: probed
[    0.699326] libphy: ag71xx_mdio: probed
[    1.326064] ag71xx ag71xx.0: no PHY found with phy_mask=00000002

Are you able to boot with AirOS, then run a "dmesg" to compare and see what chip is found?    As I recall, I think it is supposed to be an AR8032.

Joe AE6XE

ampledata, 2020-01-19:
Joe thanks for the pointers, I'll do that and get back to you with a dmesg.

ampledata, 2020-01-20:
@ae6xe dmesg from XW.v5.6.15 attached:
```
[    0.000000] Linux version 2.6.32.68 (buildd@builder) (gcc version 4.1.2) #1 Tue Mar 28 10:53:34 EEST 2017
[    0.000000] flash_size passed from bootloader = 8
[    0.000000] CPU revision is: 0001974c (MIPS 74Kc)
[    0.000000] ath_sys_frequency: cpu srif ddr srif cpu 535 ddr 400 ahb 200
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   Normal   0x00000000 -> 0x00004000
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x00004000
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 8028a440, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16256 pages, LIFO batch:3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: console=tty0 root=31:03 rootfstype=squashfs init=/init mtdparts=ath-nor0:256k(u-boot),64k(u-boot-env),1024k(kernel),6528k(rootfs),256k(cfg),64k(EEPROM) mem=64M panic=3 quiet
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 61924k/65536k available (2086k kernel code, 3540k reserved, 511k data, 136k init, 0k highmem)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:128
[    0.000000] plat_time_init: plat time init done
[    0.000000] Console: colour dummy device 80x25
[    0.000000] console [tty0] enabled
[    0.000000] Calibrating delay loop... 266.24 BogoMIPS (lpj=133120)
[    0.022000] Mount-cache hash table entries: 512
[    0.024000] NET: Registered protocol family 16
[    0.032000] bio: create slab <bio-0> at 0
[    0.034000] Switching to clocksource MIPS
[    0.036000] NET: Registered protocol family 2
[    0.037000] IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.038000] TCP established hash table entries: 2048 (order: 2, 16384 bytes)
[    0.038000] TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
[    0.038000] TCP: Hash tables configured (established 2048 bind 2048)
[    0.038000] TCP reno registered
[    0.039000] NET: Registered protocol family 1
[    0.039000] ATH GPIOC major 0
[    0.041000] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.041000] msgmni has been set to 121
[    0.041000] io scheduler noop registered (default)
[    0.065000] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.066000] serial8250.0: ttyS0 at MMIO 0xb8020000 (irq = 19) is a 16550A
[    0.067000] 6 cmdlinepart partitions found on MTD device ath-nor0
[    0.067000] Creating 6 MTD partitions on "ath-nor0":
[    0.067000] 0x000000000000-0x000000040000 : "u-boot"
[    0.069000] 0x000000040000-0x000000050000 : "u-boot-env"
[    0.070000] 0x000000050000-0x000000150000 : "kernel"
[    0.071000] 0x000000150000-0x0000007b0000 : "rootfs"
[    0.072000] mtd: partition "rootfs" set to be root filesystem
[    0.072000] 0x0000007b0000-0x0000007f0000 : "cfg"
[    0.073000] 0x0000007f0000-0x000000800000 : "EEPROM"
[    0.075000] u32 classifier
[    0.075000]     Performance counters on
[    0.075000]     input device check on
[    0.075000]     Actions configured
[    0.076000] TCP westwood registered
[    0.076000] NET: Registered protocol family 10
[    0.080000] NET: Registered protocol family 17
[... 83 lines cut ...]
[   16.153000] MAC0 MDC is 3 at line 653
[   16.153000] Setting Drop CRC Errors, Pause Frames and Length Error frames
[   16.154000] Setting PHY...
[   16.154000] Forced AR8035 PHY hardware reset on PHY setup.
[   19.170000] ADDRCONF(NETDEV_UP): eth0: link is not ready
[   19.180000] ubnt_poll: Setting timing params to min:1000 max:3000 offset:3000 tx_time: 3000 init: 5
[   19.180000] ubnt_poll: Setting aggregation params to non-txop:0 tx-op:0 scheduling:0
[   19.180000] ubnt_poll: Setting TXOP Window parameters to size:25 timeout-perc:5
[   19.180000] ubnt_poll: Module started in STA mode
[   22.197000] ADDRCONF(NETDEV_UP): ath0: link is not ready
[   27.320000] athr_gmac_ring_alloc Allocated 2048 at 0x82870000
[   27.320000] athr_gmac_ring_alloc Allocated 4032 at 0x83219000
[   27.621000] athr_gmac_mii_setup 1
[   27.621000] WASP ----> S27 PHY MDIO
[   27.621000] MAC1 MDC is 3 at line 722
[   27.621000] Setting Drop CRC Errors, Pause Frames and Length Error frames
[   27.621000] ATHRS27: resetting s27
[   27.722000] ATHRS27: s27 reset done
[   27.732000] Setting PHY...
[   28.804000] ADDRCONF(NETDEV_UP): eth1: link is not ready
[   29.169000] ATH_MAC_TIMER: MAC 0 PHY 0x04 - LINK NOW UP [link=UP speed=100Mbps duplex=full duplex carrier=0]
[   29.169000] ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   29.169000] br0: port 1(eth0) entering learning state
[   30.169000] br0: port 1(eth0) entering forwarding state
```

ampledata, 2020-01-20:
board.info:
```
board.sysid=0xe825
board.cpurevision=0x00001122
board.name=NanoBeam M5
board.shortname=N5B
board.subtype=19
board.netmodes=7
board.hwaddr=0418D60CC2E7
board.reboot=40
board.upgrade=150
board.phycount=1
board.phy.1.maxmtu=2024
board.phy.2.maxmtu=2024
radio.1.name=
radio.1.shortname=
radio.1.bus=ahb
radio.1.subsystemid=0xe825
radio.1.subvendorid=0x0777
radio.1.txpower.max=26
radio.1.txpower.min=-4
radio.1.txpower.offset=4
radio.1.ieee_modes=5
radio.1.antennas=1
radio.1.def_antenna=4
radio.1.antenna.1.id=4
radio.1.antenna.1.name=Built in
radio.1.antenna.1.gain=19
radio.1.antenna.1.builtin=1
radio.1.antenna.1.chains=2
radio.1.antenna.1.chain.1.id=0
radio.1.antenna.1.chain.1.name=Horizontal
radio.1.antenna.1.chain.2.id=1
radio.1.antenna.1.chain.2.name=Vertical
radio.1.distance_limit_km=0
radio.1.chanbw="5,8,10,20,30,40"
feature.rssi.leds=4
feature.ext_reset=0
```

ampledata, 2020-01-20:
On a whim I tried a nightly build to no avail:

```
  1234-dcdd217, r7676-cddd7b4c77
 ----------------------------------------------
root@NOCALL:~# dmesg
[    0.000000] Linux version 4.9.152 (aredn@d289f887b97a) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7676-cddd7b4c77) ) #0 Wed Jan 30 12:21:02 2019
[    0.000000] MyLoader: sysp=55755357, boardp=54161113, parts=21373774
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] SoC: Atheros AR9342 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 803aa0a4, node_mem_map 81000020
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line:  board=UBNT-NM-XW mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env)ro,7552k(firmware),256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs noinitrd
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 60696K/65536K available (2891K kernel code, 156K rwdata, 324K rodata, 268K init, 198K bss, 4840K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:535.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:40.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7144898866 ns
[    0.000011] sched_clock: 32 bits at 267MHz, resolution 3ns, wraps every 8027976190ns
[    0.008331] Calibrating delay loop... 266.64 BogoMIPS (lpj=1333248)
[    0.091165] pid_max: default: 32768 minimum: 301
[    0.096209] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.103267] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.114016] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.124542] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.131727] NET: Registered protocol family 16
[    0.138252] MIPS: machine is Ubiquiti Nanostation M XW
[    0.386889] Can't analyze schedule() prologue at 8032f6f0
[    0.404798] clocksource: Switched to clocksource MIPS
[    0.411500] NET: Registered protocol family 2
[    0.417129] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.424566] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.431403] TCP: Hash tables configured (established 1024 bind 1024)
[    0.438289] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.444525] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.451517] NET: Registered protocol family 1
[    0.456246] PCI: CLS 0 bytes, default 32
[    0.462610] workingset: timestamp_bits=30 max_order=14 bucket_order=0
[    0.477250] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.483456] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.505403] io scheduler noop registered
[    0.509581] io scheduler deadline registered (default)
[    0.515538] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.522764] console [ttyS0] disabled
[    0.546692] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 2500000) is a 16550A
[    0.555891] console [ttyS0] enabled
[    0.563306] bootconsole [early0] disabled
[... 46 lines cut ...]
[    9.910837] ip_tables: (C) 2000-2006 Netfilter Core Team
[   10.107625] ath: EEPROM regdomain: 0x0
[   10.107637] ath: EEPROM indicates default country code should be used
[   10.107641] ath: doing EEPROM country->regdmn map search
[   10.107658] ath: country maps to regdmn code: 0x3a
[   10.107664] ath: Country alpha2 being used: US
[   10.107668] ath: Regpair used: 0x3a
[   10.122594] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   10.127195] ieee80211 phy0: Atheros AR9340 Rev:2 mem=0xb8100000, irq=47
[   10.180939] kmodloader: done loading kernel modules from /etc/modules.d/*
[   11.708258] urandom_read: 5 callbacks suppressed
[   11.708267] random: jshn: uninitialized urandom read (4 bytes read)
[   28.632843] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[   28.681837] jffs2_build_filesystem(): unlocking the mtd device...
[   28.688002] done.
[... cut ...]
[   30.461706] br-lan: port 1(wlan0) entered blocking state
[   30.467175] br-lan: port 1(wlan0) entered disabled state
[   30.473090] device wlan0 entered promiscuous mode
[   30.565879] br-lan: port 1(wlan0) entered blocking state
[   30.571290] br-lan: port 1(wlan0) entered forwarding state
[   38.345011] done.
[   38.347020] jffs2: notice: (1138) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   73.791451] random: crng init done
```

ae6xe, 2020-01-20:
Give the "rocket-m-xw" image a try.   This dmesg AirOS log confirms, your device has the AR8035 PHY Ethernet chip:

[   16.154000] Forced AR8035 PHY hardware reset on PHY setup.

The  driver to support this chipset is included in the rocket-m-xw image.   Other devices where we've seen hardware revisions from AR8032 to AR8035, this image has worked.

Joe AE6XE

ampledata, 2020-01-22:
@ae6xe Looks like that worked:

```
[    0.000000] Linux version 4.9.152 (aredn@50697435f5b7) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7676-cddd7b4c77) ) #0 Wed Jan 30 12:21:02 2019
[    0.000000] MyLoader: sysp=55755357, boardp=541e1193, parts=21373774
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] SoC: Atheros AR9342 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 803aa0a4, node_mem_map 81000020
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line:  board=UBNT-RM-XW mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env)ro,7552k(firmware),256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs noinitrd
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 60696K/65536K available (2891K kernel code, 156K rwdata, 324K rodata, 268K init, 198K bss, 4840K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:535.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:40.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7144898866 ns
[    0.000011] sched_clock: 32 bits at 267MHz, resolution 3ns, wraps every 8027976190ns
[    0.008330] Calibrating delay loop... 266.64 BogoMIPS (lpj=1333248)
[    0.091165] pid_max: default: 32768 minimum: 301
[    0.096208] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.103258] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.114007] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.124525] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.131703] NET: Registered protocol family 16
[    0.138228] MIPS: machine is Ubiquiti Rocket M XW
[    0.386855] Can't analyze schedule() prologue at 8032f6f0
[    0.404779] clocksource: Switched to clocksource MIPS
[    0.411478] NET: Registered protocol family 2
[    0.417103] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.424541] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.431379] TCP: Hash tables configured (established 1024 bind 1024)
[    0.438263] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.444500] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.451491] NET: Registered protocol family 1
[    0.456220] PCI: CLS 0 bytes, default 32
[    0.462581] workingset: timestamp_bits=30 max_order=14 bucket_order=0
[    0.477226] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.483440] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.505410] io scheduler noop registered
[    0.509589] io scheduler deadline registered (default)
[    0.515542] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.522772] console [ttyS0] disabled
[    0.546702] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 2500000) is a 16550A
[    0.555900] console [ttyS0] enabled
[    0.563314] bootconsole [early0] disabled
[    0.576138] m25p80 spi0.0: found mx25l6405d, expected m25p80
[    0.588483] m25p80 spi0.0: mx25l6405d (8192 Kbytes)
[    0.593470] 5 cmdlinepart partitions found on MTD device spi0.0
[... 78 lines cut ...]
[   52.896614] br-lan: port 1(eth0) entered blocking state
[   52.901938] br-lan: port 1(eth0) entered disabled state
[   52.907797] device eth0 entered promiscuous mode
[   52.912552] br-lan: port 1(eth0) entered blocking state
[   52.917902] br-lan: port 1(eth0) entered forwarding state
[   54.686789] br-lan: port 1(eth0) entered disabled state
[   54.692215] br-lan: port 2(wlan0) entered disabled state
[   54.745713] device eth0 left promiscuous mode
[   54.750242] br-lan: port 1(eth0) entered disabled state
[   54.816877] device wlan0 left promiscuous mode
[   54.821502] br-lan: port 2(wlan0) entered disabled state
[   55.433636] eth0: link down
[   62.618477] br-lan: port 1(eth0) entered blocking state
[   62.623799] br-lan: port 1(eth0) entered disabled state
[   62.629581] device eth0 entered promiscuous mode
[   65.319045] br-lan: port 2(wlan0) entered blocking state
[   65.324449] br-lan: port 2(wlan0) entered disabled state
[   65.330344] device wlan0 entered promiscuous mode
[   65.495347] br-lan: port 2(wlan0) entered blocking state
[   65.500758] br-lan: port 2(wlan0) entered forwarding state
[   83.456252] eth0: link up (100Mbps/Full duplex)
[   83.460928] br-lan: port 1(eth0) entered blocking state
[   83.466277] br-lan: port 1(eth0) entered forwarding state
[   85.751348] random: crng init done
```

ampledata, 2020-01-22:
@ae6xe Although it looks like this is one of those devices that has the ethernet reset bug:

```
   55.433636] eth0: link down
[   62.618477] br-lan: port 1(eth0) entered blocking state
[   62.623799] br-lan: port 1(eth0) entered disabled state
[   62.629581] device eth0 entered promiscuous mode
[   65.319045] br-lan: port 2(wlan0) entered blocking state
[   65.324449] br-lan: port 2(wlan0) entered disabled state
[   65.330344] device wlan0 entered promiscuous mode
[   65.495347] br-lan: port 2(wlan0) entered blocking state
[   65.500758] br-lan: port 2(wlan0) entered forwarding state
[   83.456252] eth0: link up (100Mbps/Full duplex)
[   83.460928] br-lan: port 1(eth0) entered blocking state
[   83.466277] br-lan: port 1(eth0) entered forwarding state
```

ae6xe, 2020-01-22:
It is common to see the br-lan go through this cycle at start up.   If it occurs after the node has been booted and the ethernet port is down/up disruppting traffic, then that would be an issue.   

Do you have a test date or any other information on the packaging labels or on this device that would help to identify this new hardware rev?    This is the second device we've seen with the PHY ethernet chip upgraded from ar8032 to ar8035 chipset, but difficult to identify when this is the case.

ampledata, 2020-01-22:
@ae6xe I have about 4 devices exhibiting this behavior. Unfortunately the packaging was lost a long time ago. I'm happy to send you one to add to your collection.

ampledata, 2020-01-22:
@ae6xe 
```
[  305.135274] eth0: link down
[  305.138580] br-lan: port 1(eth0) entered disabled state
[  323.856244] eth0: link up (100Mbps/Full duplex)
[  323.860913] br-lan: port 1(eth0) entered blocking state
[  323.866259] br-lan: port 1(eth0) entered forwarding state
[  780.575082] eth0: link down
[  780.578408] br-lan: port 1(eth0) entered disabled state
[  800.336220] eth0: link up (100Mbps/Full duplex)
[  800.340884] br-lan: port 1(eth0) entered blocking state
[  800.346231] br-lan: port 1(eth0) entered forwarding state
[  803.455118] eth0: link down
[  803.458454] br-lan: port 1(eth0) entered disabled state
[  900.195335] eth0: link up (100Mbps/Full duplex)
[  900.205452] br-lan: port 1(eth0) entered blocking state
[  900.210808] br-lan: port 1(eth0) entered forwarding state
```

Related? https://dev.archive.openwrt.org/ticket/19085.html

ae6xe, 2020-01-22:
The openwrt ticket 19085 had a fix -- confirmed working.     We need to see if others can reproduce this issue as next step.   If you send one to my address at QRZ, I'll keep it in inventory for testing and resolving any model specific issues (or return if wanted).

Joe AE6XE

ampledata, 2020-01-25:
En route, no need to return. Should arrive next week.

ae6xe, 2020-02-04:
Device received.

ae6xe, 2020-02-27:
I was able to boot this device and connect via the ethernet port.  The boot up shows:

[    1.435062] ag71xx ag71xx.0: connected to PHY at ag71xx-mdio.0:04 [uid=004dd072, driver=Atheros 8035 ethernet]
[    1.446110] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode:RGMII

Was this device showing the problem?  I'm wondering if there is a hardware failure showing different results at different times?

Joe AE6XE

closed 2020-12-16 by ae6xe