aredn/aredn_ar71xx #472: New Device: GL-AR750S-EXT "SLATE"

k1ky, 2020-01-28:
Request to Add this new "Gigabit" model or review compatibility with existing code to support this device.  Featuring Gigabit Ethernet ports.
We have a couple of these devices on hand and available for testing as needed.

ae6xe, 2020-01-28:
K1KY,  telnet or ssh into the device, it is running a flavor of openwrt already.  let me know the contents of /etc/board.json .  If that doesn't exist, then /tmp/sysinfo/model .    This looks to be a slam dunk to get working in AREDN.  Appears to be minor definitions needed to configure.   It appears the default IP address is 192.168.8.1 out-of-box.

k1ky, 2020-01-28:
What is the default password and port (22?) on this device with the native openwrt installed?

k1ky, 2020-01-28:
/etc/board.json 
{
	"model": {
		"id": "gl-ar750s",
		"name": "GL-AR750S"
	},
	"led": {
		"wlan2g": {
			"name": "WLAN2G",
			"sysfs": "gl-ar750s:white:wlan2g",
			"trigger": "phy1tpt"
		},
		"wlan5g": {
			"name": "WLAN5G",
			"sysfs": "gl-ar750s:white:wlan5g",
			"trigger": "phy0tpt"
		}
	},
	"switch": {
		"switch0": {
			"enable": true,
			"reset": true,
			"ports": [
				{
					"num": 0,
					"device": "eth0",
					"need_tag": false,
					"want_untag": false
				},
				{
					"num": 2,
					"role": "lan",
					"index": 2
				},
				{
					"num": 3,
					"role": "lan",
					"index": 1
				},
				{
					"num": 1,
					"role": "wan"
				}
			],
			"roles": [
				{
					"role": "lan",
					"ports": "2 3 0t",
					"device": "eth0.1"
				},
				{
					"role": "wan",
					"ports": "1 0t",
					"device": "eth0.2"
				}
			]
		}
	},
	"network": {
		"lan": {
			"ifname": "eth0.1",
			"protocol": "static"
		},
		"wan": {
			"ifname": "eth0.2",
			"protocol": "dhcp"
		}
	}
}

k1ky, 2020-01-28:
[image]

Here is an attempted screen shot share - not sure if this will work.