aredn/aredn_ar71xx #477: 1307 OLSR Glitch?

ai6bx, 2020-02-03:
I started loading a few nodes with 1300 and then saw 1307 release have loaded a few nodes with this as well. Following a load, I note that the neighbor nodes are displaying their IP rather than their name. Dropping back to 1138 the name returns. glitch?

ae6xe, 2020-02-03:
The version of OLSR and the config settings are identical between all of
these images.   If the symptoms are repeatable, then gives something to
debug and get to the bottom of.    The symptoms have not been consistent or
easily repeatable so far.  How consistent are 1300+ images with this
behavior?

ai6bx, 2020-02-04:
Yep, I get the repeatable thing. I was loading on Rocket M5 XW units. I will do a few more tonight and see what happens or if I can find a trend.

ai6bx, 2020-03-01:
Joe,

I began loading 1375 yesterday with good results. I hit node http://10.176.139.122:8080/, same noted above,  http://10.170.193.95:8080/, backbone from Redlands to Elsinore, and http://10.176.139.122:8080/, located at Loma Linda VA. Each is a 5ghz Rocket XW. After firmware update, each shows only as an IP address from its neighbor. I did several other rockets, 3.4 and 2.4 XM models, with no problems. Each has been rebooted prior to the push and appeared to take the update fine with the exception of name display. Thoughts? 

Keith

ae6xe, 2020-03-01:
We still see this olsr issue.   In case you haven't bumped into yet,  in
Advanced configuration page, there's an option to restart olsr, check box
and save.  It doesn't actually save anything, just restarts oslr.  The
hostname should then show everywhere.   It's a pesky olsr bug...

Joe

ai6bx, 2020-03-01:
Unfortunately, there is no change when trying this.

dman776, 2020-04-07:
duplicate of #431

closed 2020-04-07 by dman776