aredn/aredn_ar71xx #491: Add support for UniFi AP Outdoor 5

ampledata, 2020-03-11:
Loaded aredn-3.19.3.0-ubnt-bullet-m-factory.bin, seems to work. Log attached.

[supportdata-NOCALL-201901301224 (2).zip](https://github.com/aredn/aredn_ar71xx/files/4321240/supportdata-NOCALL-201901301224.2.zip)

ampledata, 2020-03-11:
```node: SFWEM-SPARE-UAPO5-2F8F
model: Unknown Hardware 

!!!! UNSUPPORTED DEVICE !!!!
boardid: 0xe515
Device has not been tested. Please file a ticket with your experiences.

br-lan 04:18:D6:34:2F:8F
eth0   04:18:D6:34:2F:8F
eth0.1 04:18:D6:34:2F:8F
eth0.2 04:18:D6:34:2F:8F
tunl0  00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
wlan0  04:18:D6:35:2F:8F
wlan0-1 06-18-D6-35-2F-8F-00-00-00-00-00-00-00-00-00-00

/proc/cpuinfo
system type		: Atheros AR7241 rev 1
machine			: Ubiquiti Bullet M
processor		: 0
cpu model		: MIPS 24Kc V7.4
BogoMIPS		: 259.27
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 16
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='04:18:d6:35:2f:8f'
hsmmmesh.settings.mac2='53.47.143'
hsmmmesh.settings.dtdmac='52.47.143'
hsmmmesh.settings.node='SFWEM-SPARE-UAPO5-2F8F'
hsmmmesh.settings.config='mesh'```

ampledata, 2020-03-11:
Per: https://openwrt.org/toh/ubiquiti/unifi_outdoorplus

>  Exclusive feature is a configurable RF filter in the receive path that should allow better sensitivity in noisy environment.
> 
> Reverse engineering the control protocol of the filter is currently being worked on. Without a correctly programmed filter, the access point is deaf and therefore not usable.

ae6xe, 2020-03-12:
give this test image a try, and attach back a support dump if it loads:

https://drive.google.com/drive/folders/1_SQ7A44ENdq2-5tHz3DC4Qb8tTTvip0Z?usp=sharing

ampledata, 2020-03-13:
Thanks Joe. Doesn't seem to boot on UAPO or UAPO5. Tried sysupgrade from 3.19, tftp install of factory from a previous aredn install of 3.19, and a tftp install of factory from an airos install.

ae6xe, 2020-03-13:
What happens if you load the same target images from OpenWrt release 19.07.2?   It is a mystery why an AREDN bullet image would load, and an image targeted for this specific device would not.

ampledata, 2020-03-14:
Odd, same result. I can flash it, it'll reboot and become unresponsive. If I reboot again it reverts to the previous AirOS version (at least on the UAPO 2 GHz version). The UAPO5 isn't happy with the firmware you sent or OpenWRT.

I'm happy to send you one of each to keep and test with, but I also wouldn't want you burning more time than necessary adding support for these stretch models. I can also continue to test from here, I have a bunch of these laying around.

Let me know. Thanks for everything joe!