aredn/aredn_ar71xx #504: Mikrotik Nightly 1445 Save Changes Issue

kc2cbd, 2020-04-07:
Noticed this on a Mikrotik hAP rb-952ui-5ac2nd after changing the dtd.

Scalar value @target[1] better written as $target[1] at /www/cgi-bin/perlfunc.pm line 2059.
Scalar value @release[1] better written as $release[1] at /www/cgi-bin/perlfunc.pm line 2070.
Scalar value @_[0] better written as $_[0] at /www/cgi-bin/perlfunc.pm line 2089.
Scalar value @nums[0] better written as $nums[0] at /www/cgi-bin/perlfunc.pm line 2097.
Scalar value @nums[1] better written as $nums[1] at /www/cgi-bin/perlfunc.pm line 2097.

This also seems to be causing issues with olsrd because when I revert back to 3.20.3.0 I don't have issues with routing again. With Nightly 1445 it can't seem to route via node name, only IP addy.

On a hunch I just hit Save Changes on the Administration page on an LDF5, hAP and Basebox5 and I  get the same Scalar value errors as well.

All of my Mikrotik Nodes are now on 3.20.3.0

Thanks
Dale, KC2CBD

ae6xe, 2020-04-07:
What does “after changing the dtd” mean?

kc2cbd, 2020-04-07:
I changed from the default dtd of port 5 to port 4 & 5 on the hAP and then Save Changes.

ae6xe, 2020-04-07:
Did you remove the LAN-untagged from port 4.  If not, you need to do this for it to work.   You also, can't make this change directly in /etc/config/*.    See in /etc/config.mesh/network where there is an include of the vlan settings.  Need to edit this include file, then save settings might work.

kc2cbd, 2020-04-07:
This isn't a dtd problem...
You can just hit Save Changes in any Mikrotik device and this will manifest.

kc2cbd, 2020-04-07:
Just installed Nightly 1452 and then went to Setup and hit Save Changes and still doing it.
No changes made to the node other than updating to Nightly 1452.

[image]

r1de, 2020-04-07:
I think I recognize those variable names...

r1de, 2020-04-07:
Don't forget to close this issue.
my PR didn't do it automatically.

kc2cbd, 2020-04-07:
On a whim I did the same thing with a ubnt node and the same thing happened as well with the above  Configuration not Saved. So maybe it isn't just Mikrotik firmware that is causing it.

dman776, 2020-04-08:
fixed by #506

closed 2020-04-08 by dman776

ae6xe, 2020-04-08:
"installed nightly build" through what path?   Was the box checked "keep settings" when uploading the image?  If so, the manually edited files, incompatible with the AREDN UI, would likely have been preserved through the upgrade.   To get to 'firstboot' state, hold the reset button for 15 seconds after it is booted and live.   The node will reboot --- reverting any manually edited files and back to default 192.168.1.1 address.  Go through the UI to setup.  If you can reproduce the symptoms at that point, with out manually editing config files, then please upload a support data download, link at bottom of Administration page.   Manually editing config files requires some understanding of how the UI is working to remain compatible -- this is experimenting on your own territory.