aredn/aredn_ar71xx #507: UBNT Nanobridge M5 and ath79 Bullet Firmware

k1ky, 2020-04-07:
I noticed that the Nanobridge M5 is not listed for the ath79 Bullet firmware? I loaded 3.20.3.0-ath79-bullet-m-sysupgrade without complaint (and appears to be working) on a UBNT Nanobridge M5.  Is this an upgrade  table listing omission or should I be worried and revert back to the ar71xx-bullet version for this model?  If not, should there be additional firmware loading validation to avoid this combination?

So far - no problems or issues have been detected.

dman776, 2020-04-08:
Joe, where was the Nanobridge M5 on our testing list?

ae6xe, 2020-04-08:
Openwrt design is that every model will have a unique ath79 image.  Before many models shared the same image, and it was confusing.    Consequently, in the future, porting NB devices to ath79 target, will have a different image.     There has been some debate, e.g. here are 2 devices that are identical from the firmware perspective, but have different model names.   The openwrt decision is to continue to have different and unique images for every model regardless -- no confusion and 1 to 1 mapping all the time.

To support NBM5 on ath79, it would be best that we release this image instead of diverging from the openwrt approach.

k1ky, 2020-04-08:
Will there be any complications "reverting" back to ar71xx version or would it be best for me to wait for the dedicated ath79 release? Unfortunately this particular node is at a locked down location so I don't want to risk a crash.  The unit appears to be performing fine so far.

ae6xe, 2020-04-08:
it may revert, have to try.    Note, however, openwrt and aredn 3.20.3.0 do not completely follow this strategy of 1 image to 1 model.   nanostation-m-xw, the same image supports both M2 and M5, same for XM older models.    Although, I think the intent is still to clean this up.  looks like openwrt came to this approach after these images were created.

closed 2020-12-16 by ae6xe