aredn/aredn_ar71xx #508: Ability to "hide" local hosts on Mesh Status page

k6ccc, 2020-04-08:
I am suggesting a way to hide local services so that they do not appear on the Mesh Status page.  Here are a couple screen captures from my hAP at home that will explain this better.  First the DHCP reservations.  I have reservations for five WiFi gateways on my home WiFi network.  Currently that VLAN is not connected to any of the APs  Therefore, those IPs are not even in use - but there are reservations.  And therefore, they appear on the Mesh Status page.  Even if they were connected, there is zero reason for them to show up.
http://k6ccc.org/aredn/AREDN_DHCP-Res_page.png
http://k6ccc.org/aredn/AREDN_Local-hosts_page.png
My suggestion would be to add a checkbox for each reservation that hides that item from the Mesh Status page.  Additionally, add a checkbox that hides any attached device that has a DHCP
lease that does not have a reservation.  For example, if I plug my laptop into a node, there is generally no need for that laptop to show up on the Mesh Status page.
This would give the advantage to shortening the Mesh Status page to only useful information.
This is some related to issue #376

ae6xe, 2020-04-22:
Do you want these hostnames with DHCP reservations to be distributed across the mesh?  Thus, you can access the host remotely knowing the hostname.   Note, if the hostname is advertised across the mesh, mesh status would need some flag to know not to display -- a dilemma on how how OLSR would propagate such a flag.

k6ccc, 2020-04-22:
At least in my case, I would have no reason for the information to be advertised in any way.  Obviously if it's an advertised service, there would be little reason to hide it.  
Simple answer, if the hide checkbox is checked, OLSR does not distribute it.

Blupanthr2, 2020-04-23:
I would like to request this too.  I have a proxmox server and a static IP set for it, but I don't want it advertised on the mesh, only the virtuals that it serves.

r1de, 2020-06-24:
I think I figured out how to do this.  
So far it is working, I have a DHCP lease, I can resolve it locally _on that node_, and I can see it in the mesh list still _on that node_, but the name is no longer propagating over the rest of the mesh.  

Getting this into the GUI is a whole 'nother thing (for me at least).

Messing around with #516 (and the discussion in #538 ) gave me the idea for this issue.

r1de, 2020-06-28:
My idea for this is to have a checkbox on each DHCP lease allow the user to choose "Do not propagate this name over the mesh" or not.  
We'd just need to add a comment on the lease line in `/etc/ethers` when the lease is saved _and_ the box is checked, something like "#NOPROP".  
The line in `/etc/ethers` would now look like: `aa:bb:cc:dd:ee:ff 10.x.x.x #NOPROP`  
  
_Then_ in `/usr/local/bin/olsrd-config` we just do this:
```
if(-f "/etc/config/dmz-mode")
{
  #($lanip, $lanmask, $lanbcast, $lannet) = get_ip4_network("eth0.0");
  foreach(`cat /etc/ethers`)
  {
    ($junk, $junk, $noprop) = split ' ', $_;
    next if $noprop eq "#NOPROP";
...
```  
  
I am sure there is a much more elegant way of doing it, but that's what I came up with to check if it worked. (it _does_! :) )
  
I am still having "GUI coding block" so I am not sure when I can get this done. sorry all!

r1de, 2020-08-27:
I got over my writers block, at least for today.
Standby for upcoming PR for this issue! :tada:

PR #561 (2020-08-29): Stop certain hostname/IP's from propagating over the mesh
Allows for the node admin to choose to have DHCP leased hostnames/IP's propagated over the rest of the mesh network or not.
Defaults to allowing the hostname/IP to propagate.

The hostname/IP will *still* be resolvable from the localnode and will show up in the list of hosts on the localnode only.
This allows for selected local mesh devices to be not available over the rest of the mesh network.
ie: switches, routers, cameras, etc.

This will work immediately for *new* DHCP leases when the checkbox is selected.
For *existing* DHCP leases [... cut]
commit 8810e0ea (2020-08-29): Stop certain hostname/IP's from propagating over the mesh

Allows for the node admin to choose to have DHCP leased hostnames/IP's propagated over the rest of the mesh network or not.
Defaults to allowing the hostname/IP to propagate.

The hostname/IP will *still* be resolvable from the localnode and will show up in the list of hosts on the localnode only.
This allows for selected local mesh devices to be not available over the rest of the mesh network.
ie: switches, routers, cameras, etc.

This will work immediately for *new* DHCP leases when the checkbox is selected.
For *existing* DHCP lease [... cut]
closed 2020-08-29 by ae6xe