aredn/aredn_ar71xx #509: WAN port can't be pinged

k6ccc, 2020-04-08:
On a node that has a WAN connection, you can access the node GUI interface via the IP address of the WAN connection.  However you can't ping it.  All the other node IPs can be pinged except that one.  There are times where it would be useful for troubleshooting.

ae6xe, 2020-04-08:
The WAN port by design only accepts the following traffic, everything else is dropped.

config rule
       option src              wan
       option dest_port        2222
       option proto    tcp
       option target   ACCEPT

config rule
       option src              wan
       option dest_port        8080
       option proto    tcp
       option target   ACCEPT

config rule
       option src              wan
       option dest_port        80
       option proto    tcp
       option target   ACCEPT

It would be possible to add the following (but dtdlink example):

config rule
        option name             Allow-Ping
        option src              dtdlink
        option proto            icmp
        option icmp_type        echo-request
        option family           ipv4
        option target           ACCEPT

My Business, and I suspect many Businesses, do not enable ping, except for specific needs (watchdogs, etc.).   

Note, we can currently do the equivalent with a "psping" -- a ping on a specific port that is open, e.g. "psping <node host IP> 80".

k6ccc, 2020-04-11:
Being that in the majority cases, a node that has an active WAN port will be on somebodies LAN and therefore, behind a firewall - as opposed to directly connected to the internet.   Therefore, there is very little security risk since it still would not be pingable from the internet.

PR #544 (2020-06-25): aredn: allow wan ping
fixes: #509
commit 7b3f205f (2020-06-25): aredn: allow wan ping (#544)

fixes: #509
closed 2020-06-25 by dman776