aredn/aredn_ar71xx #516: Enhancement Request:  Add DNS Aliases

AG7GN, 2020-04-19:
Add the ability to make DNS aliases configured on a node be resolvable on nodes throughout the mesh.  With this feature, the same DHCP (or even static IP) host on an a node's LAN can be reached by names other than the host's hostname.  The DNS aliases should also be available as a dropdown when making services.

Example:  I have a Linux web server on my node's LAN.  I'd like to set up virtual servers for different purposes on that web server, each with it's own name.  The virtual server's name would have a DNS alias I'd configure on my node, and then those aliases would be resolvable on any other node on the mesh and be visible as a service on the other nodes.

Thank you,

Steve
AG7GN

solutionnuts, 2020-05-05:
This would be amazing!! Would ultimately streamline and make things more manageable.

AG7GN, 2020-05-15:
Just tried it using the nightly build and it works.  I look forward to having the aliases available in the Advertised Services URL dropdown so others on the mesh can see them.  Thanks!

Steve
AG7GN

AG7GN, 2020-06-08:
#516
@r1de On closer examination, I see that it is not working.  Has the partial implementation of the enhancement you mentioned earlier in this thread been rolled into the nightly builds?  I don't see the changes in /usr/local/bin/olsrd-config or /etc/arednsysupgrade.conf in the June 7 2020 build.

Steve
AG7GN

r1de, 2020-06-23:
no part of this is in the firmware yet.
I am having coders block with the GUI portion.

Adding the ability for OLSR to do this is trivial, but will require the user to edit a text file or 2, by hand, with the command line... :exploding_head:

AG7GN, 2020-06-27:
@r1de Suggestion:  How about a dropdown pick list containing all possible IP addresses for hosts on the LAN subnet and then underneath that, a table of 2 columns and, say, 5 rows.  The first column is a check box the second is a text field.  For the IP address selected in the dropdown, you'd enter up to 5 aliases, one in each of the text fields.  The checkbox would be used to delete an alias.  

Steve
AG7GN

r1de, 2020-06-28:
too confusing (IMO).  
As it is right now (for my testing), it looks like this:  
[image]
it'll just be a 2 choices, user input of hostname and a drop down of IP's that already have leases...
Not sure if you will be able to create an alias for _any_ available LAN IP though... I am not sure it'll work that way, maybe it will.  
Once I can get the GUI to work right I might be able to test that!!  :confounded:

AG7GN, 2020-06-29:
Good progress!  Thanks for the update.  Will you limit the number of aliases one can assign to the same IP address?

Steve

r1de, 2020-09-06:
> Will you limit the number of aliases one can assign to the same IP address?  
  
No. There is no limit.
You will be able to make aliases for DHCP leased hosts as well as for any other _non-leased_ IP's, though one does have to keep in mind that the non-leased IP's may or may not change and will have to be updated.  
  
Also, one should be able to create services based on these aliased host names! Virtual Servers here we come!

Do not get too excited yet tho, this is still a WIP.

r1de, 2020-09-07:
Had to get the aliased services to work properly, also had to add the aliases to the proper file so they would show up on the localnodes mesh status listing.  
You can't manage what you can't see! :smile:  
  
PR forthcoming after some more testing to make sure it's working.

r1de, 2020-09-08:
"test-alias" and "kg6wxc-mail" are both aliases, and you can see they show up normally.
The rest of the network see's them as normal hosts as well.
[image]
(ignore the "greyed out" names, I wont give anything away, but you might figure it out based on the other image, that change will be coming in a little bit!)

Here is what my ports page looks like now with the aliases:
[image]
  
All testing so far has been positive, lets hope it stays that way! :tada:

PR #566 (2020-09-08): aredn: Mesh host aliases
Allow for "CNAME" like aliases on the mesh.
The aliases can also be used to create propagated services using the aliased hostname.
Create new aliases on the DHCP and Ports page.
Once an alias is created it can be used to host a service, just choose the alias name in the service host dropdown.
The aliases will resolve _after_ the real hostname, so returns on pings may show a different host other than the alias, which is normal.
There is no limit for the number of aliases per IP, don't go nuts tho.
This can be used for virtual domain email servers, virtual hosts, virtual [... cut]
commit e13dd2cb (2020-09-08): aredn: allow aliased hostnames on the mesh

Allows for aliased hostnames on the mesh. One IP/Host can be assigned to multiple hostnames.
This is useful for many things including virtual hosts, virtual machines, virtual email addresses, etc.
The possibilities are actually _endless_.

Fixes #516
closed 2020-09-08 by ae6xe