aredn/aredn_ar71xx #520: Node shows as (tun) but is not configured as such.

ai6bx, 2020-04-23:
AI6BX-1-RM5-XW-Elsinore, 10.86.10.249, should show up  as a dtd with the remaining -1 nodes at this site however since upgrading it shows as (tun). Additionally, when looking at it in the ETX list it will show as (tun*1). I have tried reboots and OLSR restarts with no change. Any ideas?

Keith - AI6BX

Orv, 2020-04-23:
I've seen this behavior also.
Orv W6BI

ae6xe, 2020-04-24:
Keith, I restarted OLSR on AIBX-1-RM5-XW-Elsinore and it shows correctly now on the neighbor's mesh status.    Were you restarting OLSR on this node?    I confirmed this node was not propagating its "dtdlink.ai6bx-1-rm5-xw-elsinore" interface/hostname.    it is a duplicate of https://github.com/aredn/aredn_ar71xx/issues/431 .

Orv, 2020-04-24:
[image]

Orv, 2020-04-24:
These are Dtd links at the West LA VA.

[image]

ai6bx, 2020-04-24:
I have restarted OLSR from the advanced tab a few times when I first updated this node as it was one of the most persistent in not showing the node name and rather only the IP address. Occasionally after a reboot or restart, it will show as dtd but then reverts back.

ae6xe, 2020-08-20:
duplicate, tracked on another issues.

closed 2020-08-20 by ae6xe