aredn/aredn_ar71xx #521: CPE610 v1 boot loop after installing 3.20.3.0-ar71xx

kostecke, 2020-04-25:
I took a CPE610 which was running a nightly build (don't have the version handy) and used the AREDN WebUI Administration Page to install the recommended 3.20.3.0 sysupgrade firmware.

That left the 610 stuck in a boot loop (based on my observation of the status LEDs) of varying duration.

I've installed aredn-3.20.3.0-ar17xx-cpe610-v1-factory.bin via TFTP and was stuck in a boot loop.

I've reflashed the 610 with TP-Link Pharos [2.2.0 Build 20190228 Rel. 32857 (5553)]
via TFTP and installed aredn-3.20.3.0-ar17xx-cpe610-v1-factory.bin via the Pharos WebUI Firmware Update feature.  Again, a boot loop.

I have confirmed that the md5sum of the firmware I downloaded (http://downloads.arednmesh.org/firmware/aredn-3.20.3.0-ar71xx-cpe610-v1-factory.bin) is the same as that published at 
http://downloads.arednmesh.org/firmware/html/stable.html

What next?

kostecke, 2020-04-25:
http://downloads.arednmesh.org/snapshots/trunk/targets/ath79/generic/aredn-1509-1ace91a-ath79-tplink_cpe610-v1-factory.bin works.

Was the stable release built for the wrong target?

apcameron, 2020-04-27:
Unfortunately there seems to be a bug and I have fixed it in the nightly builds as you have discovered.

kostecke, 2020-06-15:
That's great. But the buggy version is still listed at http://downloads.arednmesh.org/firmware/html/stable.html ...

ae6xe, 2020-06-16:
This page will be correct shortly with a 3.20.3.1 point release in the pipeline.  If the point release stretches out, we'll need to change accordingly -- active discussion.

closed 2020-06-16 by ae6xe