aredn/aredn_ar71xx #522: meshchat does not sync with other nodes: Mikrotik hAP ac lite over tunnel (tun) - 3.20.3.0

mpannen1979, 2020-04-26:
I have been running down an issue that some others and I locally have been noticing with meshchat, and pings, when the other nodes are over tunnels.  This latest firmware is the first time I have ever tried tunnels, which, in general, is working for me!

However, meshchat was not syncing, and I narrowed it down to 'curl' on the mikrotik node that is either running as tunnel host/client (or both), not being able to connect to the remote node.  Curl just times out after 5 seconds, as meshchat has it do.

I don't know enough about iptables, but I think by 'accident' I found a workaround.  The file /etc/local/mesh-firewall/01-tunnels has a line:

 iptables -A zone_vpn_input -j zone_vpn_REJECT

that if I comment this line out and reboot, pings and meshchat magically start to work?!?

Can anyone help me understand what this file does, is this a bug?

I

ae6xe, 2020-04-26:
When it is failing, can you capture a support data download, link at bottom of the Administration page?   Do you by chance have the LAN configured with NAT (not the default option, you would have had to specifically change to this setting).

mpannen1979, 2020-04-26:
I have my 'workaround' active.  Would you like me to revert and do the support capture?

We have at least 3 of these hAP's that are like this, we do some tunneling with them.

We do not use the NAT setting for LAN.

mpannen1979, 2020-04-27:
I was able to replicate this on the other hAP's we use, too.  When I did the 'comment out', things starting working on those, too.

sutehk-cs, 2020-04-27:
Ran into this same issue running MeshChat from the MikroTik's internal OS. Iptables is configured to pass traffic from the LAN ports and Wifi, but not the internal system. Traffic from the internal OS failed to pass.

sutehk-cs, 2020-04-27:
Copy and paste from my local mesh group...

Ok, so here is what I had to do to get the traffic to move both ways. The firewall
has some custom rules that are added at boot for the AREDN firmware. These 
files can be found in /etc/local/mesh-firewall. The file I had to modify was
/etc/local/mesh-firewall/01-tunnels. This file adds all the custom chains and rules
pertaining to, you guessed it, the tunnels. The following two lines pertain to the custom
zone_vpn_REJECT rule where all tunnel interfaces are specified. 

iptables -A zone_vpn_REJECT -o tun+ -j REJECT
iptables -A zone_vpn_REJECT -i tun+ -j ACCEPT   #this was orignally REJECT

I'm not overly happy with this fix, but it's working. Since the Meshchat output was going
into the vpn on many different ports, none of which were in the accepted list, it was being
blocked. Not sure on how to tighten this down, or if I even need to.

mpannen1979, 2020-04-27:
Support capture with 'original'  /etc/local/mesh-firewall/01-tunnels in effect.

mpannen1979, 2020-04-27:
[supportdata-KB9OIV-1-202004261907.zip](https://github.com/aredn/aredn_ar71xx/files/4536745/supportdata-KB9OIV-1-202004261907.zip)

mpannen1979, 2020-04-27:
Here is output of iptables -S with the intact 01-tunnels
[iptables-output.txt](https://github.com/aredn/aredn_ar71xx/files/4536763/iptables-output.txt)

sutehk-cs, 2020-04-27:
The tun+ attribute to the iptables rule specifies all tunnel connections. One would have to modify them to allow the tun60 interface to pass.

mpannen1979, 2020-04-27:
Hmm, is your workaround more 'complete' then?  Will this just happen again if I add another tunnel?

mpannen1979, 2020-04-27:
I'll have to leave this one to an iptables guru!

sutehk-cs, 2020-04-27:
Joe would have a better ideas with his experience with the modified firmware. For now you may try this modification to maintain the security while allowing the necessary tunnel to be allowed.

iptables -A zone_vpn_REJECT -i tun+ !tun60 -j REJECT

Again, I am not a linux guru, this modification may not work at all.

mpannen1979, 2020-04-27:
I can replicate this without MeshChat program.  I can try to 'curl' the meshchat service or even just try to ping the next node out of the tunnel from the cli of the Mikrotik.  Pings from LAN clients (wired and wireless) plus the 'curl' are just fine.

ae6xe, 2020-04-27:
The "tun+" in iptables is a wildcard for all tun1, tun2, tun3, ...     removing the rule:
iptables -A zone_vpn_input -j zone_vpn_REJECT

may allow more traffic that is not intended,  which can open the node up for attack. 

meshchat is not AREDN released add-on, but if I recall, meshchat syncs messages with other instances via port 8080, when installed on an AREDN node, or through the already configure http server on the mesh node.    be sure when you test with curl that you do so using port 8080 reaching a mesh node.  it should accept on 80 as well, but the service is configure to use 8080 and testing should do the same.   If installed on a rasPi, test with curl on port 80 or however the Pi was configured.     (check /tmp/run/services_olsr to find all the advertised meshchat services and the port to reach them on for testing.)

If the firewall rules were broken, then it would also not be possible to reach a node across the tunnel to do a node status or mesh status on that node.  This is the same path and port numbers.   First, confirm what port the meshchat service you are trying to reach is on.  If it is 80 or 8080, then simply do "http://<node name>:8080"  to confirm you can reach it.  If it is on another port, then let's follow the path through the firewall rules (but this may be the firewall rules of the other node, that has to allow the packets to be received too.)

mpannen1979, 2020-04-27:
This is the URL that fails at the cli, with the node as is:
http://kc9mio-bfd-24-east:8080/cgi-bin/meshchat?action=messages_raw

but loads at the cli of the node if I comment out the line I mentioned, and allows meshchat to work.

the 01-tunnels file 'appears' to have a rule to allow this?

This same URL will also load on a LAN client in Chrome and return data.

mpannen1979, 2020-04-27:
It's not just that meshchat URL above, but ANY that my node tries to open that is through the tun.  MeshChat runs on the node, so this is from the viewpoint of my node with meshchat.

mpannen1979, 2020-04-27:
and icmp pings as well

mpannen1979, 2020-04-27:
sutehk-cs could probably elaborate, too, sounds like he's had the same thing.

mpannen1979, 2020-04-27:
With the new openwrt code base, did something change with iptables maybe?  total guessing...

sutehk-cs, 2020-04-27:
I've trying to get an example for you. Best I can do at the moment is the debug from MeshChat attached. I rand into this in the 3.18 and 3.19 firmwares.
[image]

sutehk-cs, 2020-04-27:
I could send messages out from the node, but not get anything back.

mpannen1979, 2020-04-27:
for instance, I cannot ping the next hAP, the 'server', from my hAP, with the way it is, i.e., KB9OIV-1<>KB9OIV-2, from KB9OIV-1 cli.

But once I comment that line out on both ends, I can.

Both of these fails have to be related?

mpannen1979, 2020-04-27:
here is some tcpdump output from the cli of the hAP, WITH the workaround:
I ran:

curl http://kc9mio-bfd-24-east:8080/cgi-bin/meshchat?action=mes
sages_raw

[tcp-dump.txt](https://github.com/aredn/aredn_ar71xx/files/4537060/tcp-dump.txt)

and 

tcpdump -i tun60 host 10.196.21.106 > /tmp/tcp-dump.txt

sutehk-cs, 2020-04-27:
This is the behavior I saw from my node.

22:06:58.347383 IP 172.31.137.173.57990 > meshchat-n3fv.local.mesh.80: Flags [S], seq 1352841749, win 28200, options [mss 1410,sackOK,TS val 1949248714 ecr 0,nop,wscale 3], length 0
22:06:58.419501 IP meshchat-n3fv.local.mesh.80 > 172.31.137.173.57990: Flags [S.], seq 1664529176, ack 1352841750, win 65160, options [mss 1460,sackOK,TS val 3846176058 ecr 1949248714,nop,wscale 7], length 0
22:06:58.419625 IP 172.31.137.173.57990 > meshchat-n3fv.local.mesh.80: Flags [R], seq 1352841750, win 0, length 0

MeshChat was sending out on port 57990 at the time to the other node on port 80. The other nodes response back was blocked and and the negotiation was reset by my node. 

Rather long tcpdump included.

[tcpdump_root.txt](https://github.com/aredn/aredn_ar71xx/files/4537089/tcpdump_root.txt)

sutehk-cs, 2020-04-27:
the zone_vpn_input chain will only allow ports:
icmp
2222/tcp
8080/tcp
80/tcp
698/udp
1978/tcp
telnet/tcp
9090/tcp
161/udp

Since the meshchatsync program uses other ports (ephemeral ports) to communicate out (allowed) the response is being blocked on the return path back into the node. Example above, N3FV's SYNC/ACK packet on port 57990. Since the meshchatsync programs from the other nodes will try to respond on these same ports they are being blocked.

Security wise, the LAN and WiFi data will be encapsulated and sent out on the allowed vpn ports and responded to encapsulated and then decoded by the node and redistributed to the appropriate destination on the correct port. Opening up the firewall rules to allow MeshChat to sync while running on the node's firmware would circumvent the vpn's security features. Had to work through it again, this is why I ended up just running MeshChat on my server. A RPi would do just as well.

mpannen1979, 2020-04-27:
Hmm, so conversely, since the tunnels aren't encrypted, and they are point to point configured  links with 'passwords' and hostnames on each end, is it a big deal if they WERE wide open?

sutehk-cs, 2020-04-27:
VPNs are encrypted. There are several protocols, some more secure than others. I'm unsure what particular protocol AREDN uses. Even so, the more ports in a firewall that are left open the higher a risk you are exposing yourself to.

ae6xe, 2020-04-27:
I’ll look at tomorrow, well later today.    Connections are tracked and there is a rule for the reply back on a dynamic port to allow.   It has happened in the past, openwrt has changed the names of some rules.   Can compare all the rules with a 3.19.3.0 node or follow the logic through the rule chains.  

Originally,  OpenSSL was in the image, but encryption and/or compression ended up being turned off on tunnels, trying to recall.  The issue was this freed up RAM because 32M devices were crashing, didn’t have enough to operate.   Then, persistent 8M flash devices didn’t have enough space to store images, so OpenSSL was removed from the image...  long story short, no encryption available on tunnels.

ae6xe, 2020-04-27:
I found the problem, but need to determine the fix.   Please compare notes on your nodes.  Do a "iptables -L -v" to inspect the output.  I usually do this command to be able to walk through the chain logic -- order is significant.    The 01_tunnel script is supposed to add entries in the designated position-order, in this case:

iptables -I FORWARD 3 -i tun+ -j zone_vpn_forward                                                                                                                                              iptables -I INPUT 5 -i tun+ -j zone_vpn_input                                                                                                                                                  iptables -I OUTPUT 3 -j zone_vpn_ACCEPT 

However, when I inspect the INPUT chain, the entry was added in position 1, not 5, for some reason.   Thus the very important rules, e.g. reply back to an established or related connection, are never encountered.  (some issue with multiple entries, but regardless, should be shifted down 5 lines.

Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination
 4253 3793K zone_vpn_input  all  --  tun+   any     anywhere             anywhere
    0     0 zone_vpn_input  all  --  tun+   any     anywhere             anywhere
    0     0 zone_vpn_input  all  --  tun+   any     anywhere             anywhere
    0     0 zone_vpn_input  all  --  tun+   any     anywhere             anywhere
    0     0 zone_vpn_input  all  --  tun+   any     anywhere             anywhere
    0     0 zone_vpn_input  all  --  tun+   any     anywhere             anywhere
 2295  163K ACCEPT     all  --  lo     any     anywhere             anywhere             /* !fw3 */
13635   11M input_rule  all  --  any    any     anywhere             anywhere             /* !fw3: Custom input rule chain */
 5220  418K ACCEPT     all  --  any    any     anywhere             anywhere             ctstate RELATED,ESTABLISHED /* !fw3 */
   34  2008 syn_flood  tcp  --  any    any     anywhere             anywhere             tcp flags:FIN,SYN,RST,ACK/SYN /* !fw3 */
    6   668 zone_lan_input  all  --  br-lan any     anywhere             anywhere             /* !fw3 */
  207 12622 zone_wan_input  all  --  eth0   any     anywhere             anywhere             /* !fw3 */
 3061 3921K zone_wifi_input  all  --  eth1.3975 any     anywhere             anywhere             /* !fw3 */
 5141 6260K zone_dtdlink_input  all  --  eth1.2 any     anywhere             anywhere             /* !fw3 */

mpannen1979, 2020-04-27:
Here is iptables from unmodified 01-tunnels from one of my hAP
[iptables-output.txt](https://github.com/aredn/aredn_ar71xx/files/4541419/iptables-output.txt)

ae6xe, 2020-04-27:
Thanks.  Confirms same behavior and wrong order.  Now do a "/etc/init.d/firewall restart" and everything should be working. (Probably need to do on both sides of the tunnel.)  This works around the issue, correct?

mpannen1979, 2020-04-27:
firewall restart breaks the tunnel, which I assume I must restart the node to fix?

Here is the iptables after firewall restart (no reboot of node yet)

[iptables-output.txt](https://github.com/aredn/aredn_ar71xx/files/4541534/iptables-output.txt)

ae6xe, 2020-04-27:
openwrt is designed to restart services and restore everything to normal operating function without rebooting the node.   The firewall restart does result in correct order of rules, from your iptables output, below -  zone_vpn_input is now the 5th entry.  

Double check, the tunnel didn't come back up?  I don't  recall it went down with my test, but regardless pings and other node to node traffic across the tunnel did work for me (which isn't the case after boot).

Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
 1084  192K ACCEPT     all  --  lo     any     anywhere             anywhere             /* !fw3 */
11488 5565K input_rule  all  --  any    any     anywhere             anywhere             /* !fw3: Custom input rule chain */
 7274 1355K ACCEPT     all  --  any    any     anywhere             anywhere             ctstate RELATED,ESTABLISHED /* !fw3 */
  236 13620 syn_flood  tcp  --  any    any     anywhere             anywhere             tcp flags:FIN,SYN,RST,ACK/SYN /* !fw3 */
 2647 2783K zone_vpn_input  all  --  tun+   any     anywhere             anywhere            
    1  1404 zone_vpn_input  all  --  tun+   any     anywhere             anywhere            
  288 20855 zone_lan_input  all  --  br-lan any     anywhere             anywhere             /* !fw3 */
  143  8642 zone_wan_input  all  --  eth0   any     anywhere             anywhere             /* !fw3 */
  563  687K zone_wifi_input  all  --  eth1.3975 any     anywhere             anywhere             /* !fw3 */
  571  708K zone_dtdlink_input  all  --  eth1.2 any     anywhere             anywhere             /* !fw3 */

mpannen1979, 2020-04-27:
I waited about 5 minutes and then just reboot the node.  The tunnel I ran your test on is the 'server' for a few clients.  My far end node just disappeared and never came back after the firewall restart, no pings.

mpannen1979, 2020-04-27:
The GUI 'thought' that the clients were connected (the cloud icon was blue), but my far node just disappeared from the web gui and that was it.

sutehk-cs, 2020-04-28:
Joe, I can confirm the placement of the iptable rules before and after restart of the firewall. After restart of the firewall, however, I could not ping any of the other nodes on the mesh. Also of note, the web GUI became unresponsive. I rebooted the node, and iptable rules reverted to original state (prior to firewall restart). No issues with the VPN tunnel during any of this, but I'm a client only. Web GUI was still giving me issues, but I want to get back to it and delete any cached info. I'll get back on that one.

ae6xe, 2020-04-28:
stall a little.  I'm doing some bigger edits to the /etc/local/mesh-firewall/01-tunnels script that is at the core of this issue.   I've found at least a couple of problems to address.  Symptoms would show up when internet is a bit unstable and tunnel link is bouncing on occasion.   Also, looks like one rule was out of order, with openwrt adding a rule and tunnel rule needed to be shifted down one position.

ae6xe, 2020-04-29:
If you are willing to test, update on a live node this file:
/etc/local/mesh-firewall/01-tunnels

with the contents of this file:
https://drive.google.com/file/d/1oeST_rqF_77yvyYgKecxPIARYRi_TGVi/view?usp=sharing

Does this fix the issue?

sutehk-cs, 2020-04-29:
Running the 3.20 firmware.

Updated the 01-tunnels file and restarted the firewall, iptables output show the original order of the rules.

Rebooted, same thing.

sutehk-cs, 2020-04-29:
Fixed my issue, here. The restart was including my backup file on init. 

Attached is the output after your update and rebooted.
[iptables-output.txt](https://github.com/aredn/aredn_ar71xx/files/4549302/iptables-output.txt)

sutehk-cs, 2020-04-29:
Just reinstalled Meshchat on the MikroTik. Started synching right away for me. Seems like you got it figured out Joe.

mpannen1979, 2020-04-29:
I dropped that into a new empty 01-tunnels file, chmod 755 for root:root

I just did a reboot, I did not mess with firewall restart.

pings, meshchat on 8080, ssh port 2222 works fine for me.

mpannen1979, 2020-04-29:
[iptables-output.txt](https://github.com/aredn/aredn_ar71xx/files/4549622/iptables-output.txt)
latest iptables for kb9oiv-1

mpannen1979, 2020-04-29:
I updates 4 tunnel nodes on 3.20.3.0 with this, and they all came back online! :)

PR #524 (2020-04-30): bugfix: aredn firewall blocking traffic when using tunnel feature
fixes #522
tested by: Matthew KB9OIV <Matthew.annen@gmail.com>
tested by: Chris K3ADA <sutehk.cs@gmail.com>

Resolves 2 issues with tunnel iptable rules.  A rule needed to be
shifted down by 1 position in chain given upgrade to openwrt 19.07.
Reload of rules was not correctly retaining chain order and creating
duplicate entries, inadvertantly blocking intended traffic.
commit b53ce1e8 (2020-04-30): bugfix: aredn firewall blocking traffic when using tunnel feature (#524)

fixes #522
tested by: Matthew KB9OIV <Matthew.annen@gmail.com>
tested by: Chris K3ADA <sutehk.cs@gmail.com>

Resolves 2 issues with tunnel iptable rules.  A rule needed to be
shifted down by 1 position in chain given upgrade to openwrt 19.07.
Reload of rules was not correctly retaining chain order and creating
duplicate entries, inadvertantly blocking intended traffic.
closed 2020-04-30 by dman776