aredn/aredn_ar71xx #526: Enhancement request: Packages repository

xJMV, 2020-05-02:
Just like #452, packages should also be able to be retrieved from the mesh or whatever URL specified in the advanced settings.

r1de, 2020-05-02:
The newer nightly builds ( > 3.20.3.0 ) should have this ability.
See the advanced config page:
[image]
You have to set each one individually (they could be in different places! :smile: )

xJMV, 2020-05-02:
Oh did not load any nodes with nightly... Sorry for that... No way to use list files ?

r1de, 2020-05-02:
not for the packages I do not think.
`opkg` uses the `/etc/opkg/distfeeds.conf` file to get the list of packages.
It's similar to apt and different than the firmware downloads.
I could be wrong though, I am not sure of the inner workings of `opkg`.

xJMV, 2020-05-02:
Lost to all this, will close the ticket. Thanks @r1de :)

closed 2020-05-02 by xJMV