aredn/aredn_ar71xx #538: Revise Help file - remove reference to Tactical Callsigns

Orv, 2020-05-31:
Under Basic Setup - Remove the following text from the help file:

It is here that you can also set a tactical name for your node. A tactical name is just another name that your node is known by. If you are familiar with DNS records, this serves a purpose similar to a CNAME record. This is helpful in an emergency deployment situation where if for example several Red Cross shelters are being linked. In addition to the normal hostname you can give each node a tactical name such as shelter1, shelter2, shelter-north, etc. Tactical names have the same restrictions as hostnames, and are accessible through DNS like the main node names are.

To set a tactical name, put a slash after the the node name then give the tactical name. For example, "ad5oo-1/shelter5".

r1de, 2020-06-03:
This goes with changing a name on a node!
There should be no "saving" of the old name, it should not become new name/old name.
"DNS" on the mesh changes to the new name anyways, why even display and replicate the old name after it's been changed?
(Sorry, this has _always_ bothered me)

ae6xe, 2020-06-20:
Tactical node names has been a feature for prior releases and as far as I know is still working in current releases of AREDN.    

r1de, there is an issue, where a name change does not get propagated and 2 hostnames may appear across the mesh network.  Thus, the new/old are showing as if there was a tactical callsign, when there is not.  This would be a defect and I think is another issue already recorded.

Orv, is this request to remove this tactical call sign feature, or is this a request that it is broken?

Orv, 2020-06-23:
It's not broken, but I've NEVER seen anyone use it.   I'll poll the users and get some opinions but I'll be 90% of them don't even know about it.

Orv, 2020-06-23:
It was a request to remove it, but based on the feedback I've gotten a) nobody knew about it and b) they think it's a good idea.    Eric and I still think implementing DNS aliases would be a better solution.

xJMV, 2020-06-23:
@Orv in fact DNS alias can do more than just tactical callsigns... so that would be an improvement and it might well replace the tactical callsign feature.

ae6xe, 2020-06-23:
Adding DNS aliases is only equal to the current tactical hostname feature if they show in mesh status.     Should all aliases be shown in mesh status?   If yes, then consider enhancing the current tactical code.  If no, then consider adding different code.

r1de, 2020-06-24:
They do show up in mesh status, if done right.  
  
Actually, though, looking at the code just now, they kind of _are_ the same thing.  
The "tactical name" just adds another `PlParam "name" "xxxxxx"` to olsrd.conf, but adding an alias for one of the hosts uses `PlParam "10.x.x.x" "hostname"`.
  
Don't see why there could not be both, or an option to set one of the hosts aliases to be the "tactical name" of the node.  
  
I have not tried making a host alias for my _node_ yet though... :thinking:

ae6xe, 2020-06-24:
It's been a while, as I re-wrote mesh status a few years back, but I think it will only show 2 hostnames in node and mesh status with the "primary/tactical" notation.   But I don't recall if it will show the right ones if you have more than 2, probably the first 2 in order only.    It may be simpler implementation to extend the tactical callsigns to achieve the desired outcome of dns aliases.  We don't need multiple implementations that achieve the same desired outcome.  Today we  have 1 'tactical' alias, just extending to have 2+.   The hostname entry, would become "primary_host_name/alias1/alias2/...".   Then touch up the code with a loop to add entries in olsr settings.  Review node and mesh status to determine how to handle showing all the aliases (or not).

r1de, 2020-06-24:
I tried setting a "host alias" for my nodes "main" ip and it ended up that my nodes name _was_ displayed as a tactical name: `new-name/old-name`, but only to the _other nodes_ on the network, I could not see it locally on the device I made the change to.
  
I think, partially at least, the 2 `olsrd.conf` options _are_ different, even tho they seem basically the same.
  
The addition I made to `/usr/local/bin/olsrd-config` to add in the host aliases was I think 3-5 lines, it was nothing.
(I am not talking about the GUI here, this is all command line still)

closed 2020-07-01 by dman776