aredn/aredn_ar71xx #539: Older Ubiquiti Bullet M2 require

spatiald, 2020-06-01:
PROBLEM:
I have a 2009-era Ubiquiti Bullet 2HP.  I didn't see an image posted for this model.  I re-imaged with the ar71 firmware and all works fine.  The early bullets were a bit confusing in marking.  On the outside, this one is labeled Bullet M2 but, in actuality, the Bullet M2 has a IC ID of 6545A-M2B ([source](https://ui.com/complicance)).  My device has a IC ID of 6545A-B2 and is therefore a Bullet 2HP.

Other info:
Bullet 2HP, 16MB SDRAM, 4MB Flash
[Datasheet](http://dl.ubnt.com/b2hp_datasheet.pdf)
FCC ID SWX-B2 (on device label)
IC ID 6545A-B2 (on device label)
[dmesg.txt](https://github.com/aredn/aredn_ar71xx/files/4711732/dmesg.txt)

Used this firmware:
File: aredn-3.20.3.0-ar71xx-ubnt-bullet-m-sysupgrade.bin
md5sum: f3ecae78b374f36bdd2af1117a279888 

RECOMMENDATION: 
Add the Ubiquiti Bullet 2HP to list of devices that AREDN firmware supports - same firmware as the Bullet Titanium.

dman776, 2020-06-01:
This device does not have enough flash storage (4MB).   The AREDN firmware file (compressed) is around 6MB.  Additionally, these ancient devices do not support the 802.11n protocol (only a/b/g).

closed 2020-06-01 by dman776

spatiald, 2020-06-01:
Thanks dman776 and no worries...thought I would let you know that this ancient device IS working with 3.20.3.0 (ar71xx target) and supports b/g/n.  Not sure my comments were interpreted correctly.  While it is definitely not an ideal device (being so ancient), it does work and is joined to my AREDN local mesh.  It flashed with the firmware mentioned above; both with factory and sysupgrade.  It did not flash with successfully with ath79 target, most likely due to the reasons you mentioned above.

dman776, 2020-06-01:
Ah.  Got it.   Can you attach a supportdata file from the node?

Thanks,
Darryl

spatiald, 2020-06-01:
Apologies...I saved it off but never attached :) 

[supportdata-KC3FRC-bu2o-outside-202006011319.zip](https://github.com/aredn/aredn_ar71xx/files/4712088/supportdata-KC3FRC-bu2o-outside-202006011319.zip)

dman776, 2020-06-01:
so, the board_id pulled from hardware is id:   0xe202 which is definitely a Bullet M2 HP.  You are on a supported device.

spatiald, 2020-06-01:
Thus, the mystery...it's up to you if you want to list it in the model listing.  

Confusing to me at least...it is labeled M2 and the IC ID is definitely that of a Bullet 2HP (6545A-B2). According to UI's documentation, the IC ID for the Bullet M2 should be 6545A-M2B.  BTW I corrected the above "fat-fingered" link from UI.