aredn/aredn_ar71xx #540: CPE220 - Boot Loop on 3.20.3 and 3.19.3

berniemcintosh, 2020-06-05:
Router:  CPE220 V3.0  with the (UN) designation.

AREDN stable releases 3.19.3 and  3.20.3 fails to load factory image using either TFTP or from the OEM web upload page. The router appears broken - every now and again all LEDs flash then nothing.  Believed to be continually rebooting.

Able to TFTP recover to OEM firmware.

On OpenWRT website there is no stable supported release yet for this router. Loaded the latest generic ATH79 factory snapshot (cef0f85a2271606d9a9be2e81494a3834b99e1e7f85dfe9f79fdbf54234d7a0d - June 4th). This worked.

Then from CLI did a sysupgrade with AREDN 3.20.3 sysupgrade image which worked.

73 Bernie

ae6xe, 2020-06-06:
KK4ZUZ found the root cause and is working on the fix.

berniemcintosh, 2020-06-06:
Brilliant. Let me know and I will test. I have an unopened CPE220 (UN) yet to be flashed.

apcameron, 2020-06-09:
I have created #541 to Address this
Once it is approved the fix will appear in the Nightly build.