aredn/aredn_ar71xx #549: Feature Request: add Signal/Noise/Ratio values to sysinfo.json

xJMV, 2020-07-18:
It would be interesting to add the Signal/Noise/Ratio displayed on the node status page to the sysinfo.json output as 3 different values.

Main use for us would be to monitor those values with Zabbix (or any other soft) and graph them. It could help to troubleshoot and fix links with timely noise level (we face it with 900Mhz actually, noise raise to the roof once in a while). We look to collect data so we can file a complain and provide raw data to inspectors.

In the long term, it may also be helpful to see signal differences between the season (snow, ice, foliage...) OR with the evolution of other networks around and adapt nodes accordingly.

dman776, 2020-07-18:
try:   http://localnode:8080/cgi-bin/api?chart=realtime

xJMV, 2020-07-18:
Interesting... is the API documented somewhere ? I do not see it in the readthedocs...

ab7pa, 2020-08-19:
https://arednmesh.readthedocs.io/en/latest/arednHow-toGuides/devtools.html#add-link-information 
These documented RF link values are what I've used regularly for Zabbix monitoring.

ae6xe, 2020-08-20:
Issue is addressed, closing.

closed 2020-08-20 by ae6xe